contiki-os/contiki #1893: [CC1310] 433MHZ settings

mynavy, 2016-10-19:
Hello , 

CC1310 433MHZ has been added to the latest version of SmartRF Studio 7 (V2.4.3).
Currently contiki doesnt support 433MHZ settings.
Is there any future request for this kind of settings?

jonnteolsson, 2016-10-29:
Most of the bands in IEEE802.15.4g is supported, using DOT_15_4G_CONF_FREQUENCY_BAND_ID in dot-15-4g.h 

We are currently looking at the actual RF settings and will most probably make updates, but basic support should work.

Lir10, 2016-10-30:
Thanks for the answer , 

We've already tryied to add 433MHz support to dot-15-4g.h , testing with CC1310 Rev A we got poor results with contiki ( Short range & poor RSSi -90dbm) but great results with SmartRF-Studio ( -110dbm RSSI )

We made tests with data rate of 50kbps with the same settings from SmartRF-Studio.
We hope that with the new upcoming 433MHz support update the results will be different

g-oikonomou, 2016-10-30:
Can you try with NullRDC and report whether you are seeing the same RSSI situation? This sounds very similar to #1659

Lir10, 2016-10-31:
We've already tryied with NullRDC - same results.
We also tryied the solution given in #1659  but got the same results ( short range & poor RSSI)

g-oikonomou, 2016-11-07:
@Lir10 I'll look at the settings exported by SmartRF studio for 433MHz. In terms of RSSI, we currently use a very conservative -90 threshold: We assume no traffic if the RSSI is below that.

You can change this by defining a value for `PROP_MODE_CONF_RSSI_THRESHOLD` in your `contiki-conf.h` or `project-conf.h`

This may also be related with overrides, this is something we are looking at internally.

Lir10, 2016-11-08:
@g-oikonomou We've changed `PROP_MODE_CONF_RSSI_THRESHOLD` to -113 but didnt saw any difference.

g-oikonomou, 2016-11-13:
Hi @Lir10 

433MHz is not a .4g band (not to my knowledge anyway), hence the lack of the corresponding block in `dot-15-4g.h`.

You can manually hack it in, like so:

```
#elif DOT_15_4G_FREQUENCY_BAND_ID==DOT_15_4G_FREQUENCY_BAND_433
#define DOT_15_4G_CHANNEL_MAX          1
#define DOT_15_4G_CHANNEL_SPACING    200
#define DOT_15_4G_CHAN0_FREQUENCY 433920
#define PROP_MODE_CONF_LO_DIVIDER   0x0A
```

Then choose it in your `contiki-conf.h` or `project-conf.h`.

You will need to select values for `DOT_15_4G_CHANNEL_MAX` and `DOT_15_4G_CHANNEL_SPACING`. The driver will then do everything else for you.

We have intentionally focused on .4g-compliant operation for the prop-mode driver and hence we have only defined bands specified in the .4g standard. I am not foreseeing adding off-the-shelf defines for the 433MHz band in the future. Any particular reason why you do not want to use the 470MHz band?

As for the range and RSSI situation, I am pretty convinced this has to do with #1659 and not with the particular band.

Lir10, 2016-11-14:
Hi @g-oikonomou ,

Thanks for the answer.

We designed our custom boards antenna's specifically to 433MHz band  , currently our product is not working properly because of the poor range results we get with contiki as for 433MHz band (With SmartRF-Studio the results are way better). 

Changing our custom boards antenna's specifically to 470MHz band will be a major pain to us  (it's the last solution to our problem).

As for the changes in  `dot-15-4g.h` - all our tests were made with the exact changes and still got poor range results + RSSI . 

We currently dont have any clue how to continue with our project if we still want to use CC1310 with Contiki as for 433MHz band. 

We guess that we'll need to write our own prop-mode driver for 433MHz band, but still it's only a guess.

jonnteolsson, 2016-12-05:
We are preparing a PR with updated 400Mhz settings.

Lir10, 2016-12-06:
We are looking forward to it, thanks

g-oikonomou, 2016-12-15:
There should be a good starting point in #1932

Lir10, 2017-01-15:
Tryied the above suggestions but still got same results

xioabudian, 2017-06-19:
I have met the same problem,but I don't know why it is so

gani-koduri, 2018-10-28:
Any one has working solution for this ?
I want to change frequency of operation to 433 MHz for my custom module running cc26xx-web-demo application. 
I dont see any macros mentioned in thread related to frequency band in project-conf.h or contiki-conf.h files. I am not using contiki-ng yet. 

Please share the changes to be done for this requirment of changing  RF frequency of operation ?

xioabudian, 2018-10-28:
Yes,contiki-os does not support 433MHZ configuration.If you want to use 433MHz,you should change cc26xx-cc13xx driver configuration.First,you use TI SmartRF Studio software to get 433MHz configurations smartrf-setting.c and smartrf-setting.h instead of old files.
Then set work environment to DOT_15_4G_FREQUENCY_BAND_450 in dot-15-4g.h or create DOT_15_4G_FREQUENCY_BAND_433.I confirm that i get right
configurations on cc1310 by this way.