contiki-os/contiki #1894: uniqueness issue with zoul board while running more than one client

amar22, 2016-10-19:
Hello,
Issue:
we are working on lwm2m with zolertia-zoul-board ,while working with more than one board,we found that all the devices come up with same node id(as it is taking first four bytes of mac address which is same for all zoul board) .so we were not able to control more than one device at a time.
let us consider two board having mac address  **00:12:61**:15:a0:5e and **00:12:61**:15:b0:5f, from these two mac address we will get same node id (ZOLERTIA RE-MOTE Pl0**01261**)for both  board which makes us difficult to access more than one device.
Solution:
To overcome this we found out a solution : consider all the bytes of mac address which will be unique for all board this makes us easy to access more than one device at a time. To consider all bytes of mac address along with board name  go to file( /contiki/apps/oma-lwm2m/ lwm2m-engine.c) ,look out for the following function lwm2m_engine_init() ,change the len variable to  len = sizeof(client) - 19 and change the size of endpoint array size from 32 to 36.
now let us consider same boards having mac address  00:12:61:15:a0:5e and 00:12:61:15:b0:5f which will have the node id's ZOLERTIA RE-MOTE 00126115a05e and ZOLERTIA RE-MOTE 00126115b05f respectively .
This is how we can overcome the above mentioned issue.

alignan, 2016-10-19:
Changed the subject title as it is not related to the Zoul module, but to the LWM2M implementation

joakimeriksson, 2016-10-19:
This solution will only work if you allow slightly longer endpoint prefix. But why did you need to change the 13 to 19: 13 chars if for 6 * 2 chars MAC and then a zero at the string end? A fix could be to allow the user to define a prefix of any size and set all the other limitations based on that or always go for a long prefix. 32 bytes in total was just picked to get farily limited size of the required buffer/memory used but not too small to enable the client to have a meaningful name.

amar22, 2016-10-20:
@joakimeriksson 
i think you have missed the string "?ep=" which (will get added in the endpoint buffer)will take 4 character space .Because of this MAC address get push out from the endpoint buffer.

If we will take 18 char instead of 13 
```
`if(len > sizeof(client) - 18) {
    len = sizeof(client) - 18`
```
(14 byte for client name ,12 byte for MAC & 4 byte for string "?ep=", 1 byte for '\0') ,we have enough space for MAC and string called "?ep=".