contiki-os/contiki #1899: Bad RPL route with Security LLSEC on 6lbr cc2650 and sensortag

pierre-microlide, 2016-10-21:
Hi,
we started develop our contiki project one month ago to adapt wireless sensor on our data logger.
Border router : 6lbr on a linux board and a launchpad cc2650 with slip-radio.
Everything works fine with 10 sensors (sensortag) : RPL route is good.

1) To have different wireless networks on different border routers we want to use the security key with LLSEC : I don't know if it's the good way ? 

2) we program a security key on the border router and sensortag : it works but the RPL route is very bad. We loose sensors regularly, they want to connect directly on the border router instead of using a sensor node : The route between two sensors is very rarely effective ==> The mesh doesn't work correctly .... If we only remove security all is ok....
Configuration for LLSEC =
#undef NETSTACK_CONF_FRAMER
#define NETSTACK_CONF_FRAMER noncoresec_framer
#undef NETSTACK_CONF_LLSEC
#define NETSTACK_CONF_LLSEC noncoresec_driver
#undef LLSEC802154_CONF_ENABLED
#define LLSEC802154_CONF_ENABLED 1
#ifndef NONCORESEC_CONF_SEC_LVL
#define NONCORESEC_CONF_SEC_LVL 6
#endif
#undef AES_128_CONF
#define AES_128_CONF aes_128_driver
#define LLSEC_ANTIREPLAY_ENABLED 1
#define LLSEC_REBOOT_WORKAROUND_ENABLED 1

Thank's in advance for help.
Pierre

kkrentz, 2016-10-28:
Could you please try if #1703 resolves your problem?

pierre-microlide, 2016-11-29:
Hello kkrentz, excuse me for delay but i was out of office since 3 weeks.
I looked adaptativesec but i have a question : We use 6lbr on our linux board to manage sensors.
Can we use adaptativesec for the sensors and noncoresec for the border router with 6lbr ?
Thank's

naguirre, 2016-12-13:
@kkrentz does #1703 supports cc26xx platforms ?

g-oikonomou, 2016-12-15:
@naguirre We observed some stack overflow problems with LLSEC enabled in our custom CC2650-based board. Try reserving more RAM for the stack in the linker script and see if that will help you.

amitbhanja, 2018-05-01:
Hello guys, 
Did you find a way to resolve this issue using noncoresec?

Regards,
Amit

pierre-microlide, 2018-05-07:
Hello,
finally, i didn't use noncoresec in my project.... And my différents tests was not good.
Regards
Pierre