contiki-os/contiki #1912: CC2538: deadlock in cc2538_ccm_star_driver

kkrentz, 2016-10-28:
The `cc2538_ccm_star_driver#aead` never returns when I call it with `a_len = 0` and `m_len = 0`. Probably this is easy to fix for somebody who knows that code.

bthebaudeau, 2016-10-28:
Are you talking about encryption or decryption (i.e. forward or not)?

For encryption, if both `a_len` and `m_len` are `0`, then there is both no additional authenticated data and no message to authenticate and encrypt, which means asking the driver to work upon nothing. This case is not supported by `ccm_auth_encrypt_start()`. IIRC, this is a hardware limitation.

Why should this case be supported? What would be the expected behavior, were it to be supported?

bthebaudeau, 2016-10-28:
Another possibility would be to make `aes_auth_crypt_start()` return an error code instead of hanging in that case, but the `ccm_star_driver` API does not support returning error codes, so that would mean handling it with a reboot, and the AES/CCM drivers currently assume that only legal arguments are used, so that would mean a change in this rule.

But again, why would one need to authenticate or encrypt nothing?

kkrentz, 2016-10-29:
Normally, this case is not needed. However, in a custom protocol, I generate a MIC over the nonce, without authenticating additional data. As per RFC 3610, this should work and return a MIC.

bthebaudeau, 2016-10-30:
I'll see what can be done. Perhaps another hardware mode can be applied to the nonce in this case in order to get an equivalent behavior. If not, a software fallback will have to be used in this case.