contiki-os/contiki #1915: cc1310 launchpad contiki web sensor demo not working

dvsharma, 2016-11-01:
Hi All,

I have been trying to setup cc1310 launchpad based Web sensor demo for past few days now. I have followed all instructions properly (or at least that is what I think). I am not sure what I am missing, but not able to get sensors detected on 6lbr webpage. Please someone help?

I am attaching all logs:

6lbr webpage snapshot
6lbr.conf
leaf node serial logs (channel no etc is correct)
ifconfig 
I am on latest commit.

Important notes:

Serial communication seems okay. SLIP radio seems to be detecting fine.
My HW is good because I have tested 868Mhz with CCS Rx-Tx example code.
Mostly following this: sunmaysky.blogspot.sg/.../CC1350

Thanks.
[logs.zip](https://github.com/contiki-os/contiki/files/563470/logs.zip)

g-oikonomou, 2016-11-07:
Have you set the correct 6lowpan prefix on 6lbr (should be fd00:: if you are using latest Contiki)? I cannot find a link right now, but you will be able to find more info on the mailing list.

dvsharma, 2016-11-07:
Yes- Thanks I could get it to work with fd00::. Few questions I had though as I see it very unstable. Once in a while it connects, rest of the times I fail to read data from device.

- Is there one good commit that I can rely on working with cc1310? contiki3.0 is the right one? I can see all TI latest updates did not make it to contiki 3.0?
- I have 433Mhz cc1310 launchpads. I am not sure if that is the reason for instability. I saw a thread running on 433Mhz support. I am not sure if that is in yet.
- I was trying to setup sniffer, but I failed there too because contiki tree doesn't seem to have code or it doesn't seem to build properly. I had to get the code from some other site to get it to build.

You guys doing great job! Appreciate it. Looking forward to seeing stable contiki on my platform which I have not achieved yet.

g-oikonomou, 2016-11-07:
Sorry, I've never tested 433MHz at all. See the other thread on that one, we _are_ looking at things.

Has your LP actually been designed/sold for 433MHz (I'm thinking matching components here)?

dvsharma, 2016-11-09:
Would someone know about 6lbr instability that i see below? There are two global addresses that pop up after sometime of running application. After this sensor node keeps sending data, but 6lbr web page is screwed up.

dvsharma@ubuntu:~$ ifconfig
br0 Link encap:Ethernet HWaddr 00:0c:29:54:12:a7
inet6 addr: bbbb::6dd0:b916:9276:4fe6/64 Scope:Global
inet6 addr: bbbb::20c:29ff:fe54:12a7/64 Scope:Global
inet6 addr: fe80::20c:29ff:fe54:12a7/64 Scope:Link

Why there are two global addresses and where did this new ipv6 address come from?

g-oikonomou, 2016-11-13:
It appears the problem reported in the OP here is sorted out.

For the remaining questions:

* 433MHz operation: #1893 
* Sniffer is confirmed to work with CC1310
* Latest TI updates: #1924 
* As for the latest question about 6lbr instability and v6 addresses, I am not seeing anything wrong at all in this `ifconfig` output.

I'm closing this.

closed 2016-11-13 by g-oikonomou