contiki-os/contiki #1916: No rule to make target `obj_c64/logscr.S'

buttha, 2016-11-01:
Hi, I'm trying to compile contiki 3.0 for commodore 64. This is the error:

MacBook-Pro:contiki buttha$ make -C examples/webserver/ disk  TARGET=c64 HTTPD-CFS=1
using saved defines 'CONNECTIONS=4,WITH_LOGGING,WITH_BOOST,WITH_PFS'
make: *** No rule to make target `obj_c64/logscr.S', needed by `obj_c64/logscr.o'.  Stop.

oliverschmidt, 2016-11-01:
It work in general. See i.e. https://travis-ci.org/contiki-os/contiki/jobs/172110359#L3148

buttha, 2016-11-02:
thank you, I will try to understand what's wrong in my envirovment

regards
Andrea

greg-king5, 2016-11-02:
It might be the upper-case "`.S`" suffix.  List the contents of subdirectory "`platform/c64/lib`".  See if the suffixes of the assembly files are shown as upper-case.  If they are lower-case, then try to rename them to shifted 'S'.

buttha, 2016-11-03:
thank you, but no, this is not the problem

```
MacBook-Pro:contiki buttha$ ls platform/c64/lib/
exec.c		lseek.c		pfs-dir.c	pfs_remove.S	pfs_write.S
logscr.S	pfs-dir-asm.S	pfs.S		pfs_seek.S
```

maybe something is missing in my system. I will try in a linux virtual machine

closed 2017-06-16 by buttha