contiki-os/contiki #1922: CC2538-rf.c  stuck in a Infinite loop

GregCJam, 2016-11-04:
In the file cpu/CC2538/devCC2538-rf.c 

The function channel_clear() there is the following code

while((REG(RFCORE_XREG_RSSISTAT) & RFCORE_XREG_RSSISTAT_RSSI_VALID) == 0);

I found that the code can occasionally never return from this statement. 

In my application I am using a very aggressive power saving strategy and that may be the cause of this problem. It will run for 1 to 2 days then get stuck in a infinite loop

A simple fix 

 uint32_t count = 0;

#define WAIT_TIME 0x10000

  while((REG(RFCORE_XREG_RSSISTAT) & RFCORE_XREG_RSSISTAT_RSSI_VALID) == 0)
  {
	  count++;
	  
	  if ( count > WAIT_TIME )
	  {

		PRINTF("clr busy\n");
		return  CC2538_RF_CCA_BUSY;
	  }
  }

Regards

Greg Jamws

alignan, 2016-11-04:
Thanks for reporting, I think I have seen this happening from time to time, what application were you using?

I think a better solution would be to copy the macro below from the `cc1200` driver:

````
#define BUSYWAIT_UNTIL_STATE(t) \
  do { \
    rtimer_clock_t t0; \
    t0 = RTIMER_NOW(); \
    while(((REG(RFCORE_XREG_RSSISTAT) & RFCORE_XREG_RSSISTAT_RSSI_VALID) == 0) && RTIMER_CLOCK_LT(RTIMER_NOW(), t0 + (t))) {} \
    if(!(RTIMER_CLOCK_LT(RTIMER_NOW(), t0 + (t)))) { \
      printf("RF: Timeout exceeded in line %d!\n", __LINE__); \
    } \
  } while(0)
````

GregCJam, 2016-11-05:
Hi Antonio
I am designing a sensor that monitors gas concentrations ( Carbon Monoxide,
Methane , S02 ) . It sends a UDP packet every 5 seconds with the
measurement.

We found it would get stuck in a infinite loop on random occasions ( 1 hour
to 2 days)

I will give your suggestion a go.