contiki-os/contiki #1923: Running Event timer in Z1 motes in cooja

Raziur09, 2016-11-06:
Hi, I am new to contiki and i am running the mesh network for Rime in cooja. I just added the timer instead of button sensors inside the while loop like below:

/*
etimer_set(&et, 10*CLOCK_SECOND);
PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&et));
*/

Motes are supposed to send a packet with period of 10 seconds but somehow they are sending packets every 20 seconds. Whatever value I provide is just been multiplied by two. Can anyone suggest a reason for this please ? Moreover, apart from Z1 motes all other kinds of motes works as expected.

alignan, 2016-11-06:
There was a pull request merged a time ago solving something related, are
you using the latest Contiki master?

Raziur09, 2016-11-06:
Thanks for replying. I just downloaded Contiki3.0 last month and using the files under Contiki folder not Contiki3.0.

alignan, 2016-11-06:
And have you updated your working copy? Try to do a "'git fetch" and "'git
pull", and also a "'git submodule update"

--Antonio

Raziur09, 2016-11-06:
Hey thanks a lot. You already solved the issue. I did not run "git fetch" and "git pull" before. Thanks again for helping.

closed 2016-11-06 by Raziur09