contiki-os/contiki #1925: Latest sensniff doesn't work

chenek, 2016-11-07:
I pull the latest sensniff and test it. It doesn't work on my CC2650 LaunchPad and CC2650STK. Any idea?

g-oikonomou, 2016-11-07:
@chenek I'll test. What's the symptom? Are the host-to-device commands working at all?

arurke, 2016-11-07:
Is it platform specific @chenek ? I used it on a cc1310 launchpad recently - if I recall correctly there was some pitfalls in the new sensniff version; a different default baudrate and you need to specify channel via cmd.

chenek, 2016-11-07:
When I open wireshark, there is nothing coming. If I uses CC2531EMK, I can see there are lots of over the air messages. I don't test with CC1310 LanuchPad. I just test it with CC2650LP and CC2650STK.

oharakr, 2016-11-07:
I am experiencing almost the same issue. The difference is that when I open Wireshark, I only see Malformed Packets. 2bytes, 0x00 0x00

g-oikonomou, 2016-11-07:
Firstly, make sure you have the latest version of the python script. Make sure you run it with a `-b` that corresponds with your device's baud rate.

If you are seeing no traffic, use the python script's `c` interactive command to read the current channel being sniffed. Make sure the device firstly responds to that in the first place and secondly is sniffing the right channel. 

You can also run the python script with `-D INFO` and it will tell you whether it's seeing traffic in the first place. This will help determine whether traffic is not being captured vs not being displayed on WS.

g-oikonomou, 2016-11-07:
I should say, default baud rate for all CC13xx/CC26xx devices is 460800, which is also the default for the latest version of the python script.

oharakr, 2016-11-07:
Yes, I tried the latest script version and now it is working. Sorry that I didn't realized my script was outdated, it was working properly with the CC2530 so I never updated it.

g-oikonomou, 2016-11-07:
So, @oharakr, are you saying that you succeeded with CC2650LP? Or was it some other platform?

oharakr, 2016-11-07:
Yes, that's right, I succeeded with the CC2650LP. Thank you again.

chenek, 2016-11-08:
@g-oikonomou After I update sensniff.py, it works now. Thanks for the help.

closed 2016-11-08 by chenek