contiki-os/contiki #1943: IEEE 802.15.4e TSCH (TimeSlotted Channel Hopping) and CC2538dk

lawpeter, 2016-11-23:
Hello, 
recently there has been TSCH merged into Contiki (contiki/core/net/mac/tsch/README.md)

I would like to try it on CC2538dk platform but haven’t been able to build it due to some compilation errors ( below).
I hope someone can give me solution for it. 

../../../core/net/mac/tsch/tsch-slot-operation.c: In function '**tsch_tx_slot**':
../../../core/net/mac/tsch/tsch-slot-operation.c:472:95: error: 'RADIO_DELAY_BEFORE_TX' undeclared (first use in this function)
           TSCH_SCHEDULE_AND_YIELD(pt, t, current_slot_start, tsch_timing[tsch_ts_tx_offset] - RADIO_DELAY_BEFORE_TX, "TxBeforeTx");
                                                                                               ^
../../../core/net/mac/tsch/tsch-slot-operation.c:288:51: note: in definition of macro 'TSCH_SCHEDULE_AND_YIELD'
     if(tsch_schedule_slot_operation(tm, ref_time, offset - RTIMER_GUARD, str)) { \
                                                   ^
../../../core/net/mac/tsch/tsch-slot-operation.c:472:95: note: each undeclared identifier is reported only once for each function it appears in
           TSCH_SCHEDULE_AND_YIELD(pt, t, current_slot_start, tsch_timing[tsch_ts_tx_offset] - RADIO_DELAY_BEFORE_TX, "TxBeforeTx");
                                                                                               ^
../../../core/net/mac/tsch/tsch-slot-operation.c:288:51: note: in definition of macro 'TSCH_SCHEDULE_AND_YIELD'
     if(tsch_schedule_slot_operation(tm, ref_time, offset - RTIMER_GUARD, str)) { \
                                                   ^
../../../core/net/mac/tsch/tsch-slot-operation.c:479:11: warning: implicit declaration of function 'US_TO_RTIMERTICKS' [-Wimplicit-function-declaration]
           tx_duration = TSCH_PACKET_DURATION(packet_len);
           ^
../../../core/net/mac/tsch/tsch-slot-operation.c:501:102: error: 'RADIO_DELAY_BEFORE_RX' undeclared (first use in this function)
                   tsch_timing[tsch_ts_tx_offset] + tx_duration + tsch_timing[tsch_ts_rx_ack_delay] - RADIO_DELAY_BEFORE_RX, "TxBeforeAck");
                                                                                                      ^
../../../core/net/mac/tsch/tsch-slot-operation.c:288:51: note: in definition of macro 'TSCH_SCHEDULE_AND_YIELD'
     if(tsch_schedule_slot_operation(tm, ref_time, offset - RTIMER_GUARD, str)) { \
                                                   ^
../../../core/net/mac/tsch/tsch-slot-operation.c: In function 'tsch_rx_slot':
../../../core/net/mac/tsch/tsch-slot-operation.c:662:89: error: 'RADIO_DELAY_BEFORE_RX' undeclared (first use in this function)
     TSCH_SCHEDULE_AND_YIELD(pt, t, current_slot_start, tsch_timing[tsch_ts_rx_offset] - RADIO_DELAY_BEFORE_RX, "RxBeforeListen");
                                                                                         ^
../../../core/net/mac/tsch/tsch-slot-operation.c:288:51: note: in definition of macro 'TSCH_SCHEDULE_AND_YIELD'
     if(tsch_schedule_slot_operation(tm, ref_time, offset - RTIMER_GUARD, str)) { \
                                                   ^
../../../core/net/mac/tsch/tsch-slot-operation.c:676:38: error: 'RADIO_DELAY_BEFORE_DETECT' undeclared (first use in this function)
       rx_start_time = RTIMER_NOW() - RADIO_DELAY_BEFORE_DETECT;
                                      ^
../../../core/net/mac/tsch/tsch-slot-operation.c:762:19: warning: implicit declaration of function 'RTIMERTICKS_TO_US' [-Wimplicit-function-declaration]
                   &source_address, frame.seq, (int16_t)RTIMERTICKS_TO_US(estimated_drift), do_nack);
                   ^
../../../core/net/mac/tsch/tsch-slot-operation.c:776:73: error: 'RADIO_DELAY_BEFORE_TX' undeclared (first use in this function)
                   packet_duration + tsch_timing[tsch_ts_tx_ack_delay] - RADIO_DELAY_BEFORE_TX, "RxBeforeAck");
                                                                         ^
../../../core/net/mac/tsch/tsch-slot-operation.c:288:51: note: in definition of macro 'TSCH_SCHEDULE_AND_YIELD'
     if(tsch_schedule_slot_operation(tm, ref_time, offset - RTIMER_GUARD, str)) { \
                                                   ^
make: *** [obj_cc2538dk/tsch-slot-operation.o] Error 1
rm obj_cc2538dk/startup-gcc.o

alignan, 2016-11-27:
What example are you trying? I just tested the `examples/ipv6/rpl-tsch` and `examples/rime-tsch` and both are compiling OK.

lawpeter, 2016-11-28:
Thank you for your reply, actually I used wrong version. It's working well now. 
Could you tell me how to ping nodes in rpl-tsch networks ? [I tried with apps (ping6 or shell), it does not work]. 
Thank you.

lawpeter, 2016-11-28:
Anyway, thank you for your book (https://github.com/alignan/IPv6-WSN-book). 
It's very useful for newbie like me.

alignan, 2016-11-28:
Thanks!

I think if you flash the `rpl-tsch` and the `border router`, you would be able to ping the device from your host, no need to flash an additional application, as the `border router` will "forward" the ping requests and replies.

closed 2016-12-01 by alignan