contiki-os/contiki #1945: Contiki ND6 and default router (RA)

CogPy, 2016-11-23:
My setup is not using RPL at the moment (no decent linux support available to my knowledge).

Using one device with radvd as RA daemon works fine. If a second devices is used in the same /64 sub-net things start to go wrong. Technically, a simple neighbour-discovery (ND) ICMP for a different host that is not online, i.e., sleepy-nodes, will have the default route removed.

Is this a bug or a feature?

CogPy, 2016-11-23:
The issue was, that the *lowpan0* device had two different link-local, i.e., fe80::x, addresses. This then caused problems in `nbr_table_add_lladdr(...)`, as it uses the same link addresses.

Lesson: only use one fe80::/64 address on a lowpan0 device.

CogPy, 2016-11-24:
The 64-LSB of the iface's IPv6 addresses shall not be different and must be equal to the link address, i.e., the hardware provided address.

CogPy, 2016-11-24:
Seems to problem goes deeper; a NS from the default router for a different host can render the default-router unreachable. This might fix it:

```diff
diff --git a/core/net/nbr-table.c b/core/net/nbr-table.c
index 4893a99c..c7bc253 100644
--- a/core/net/nbr-table.c
+++ b/core/net/nbr-table.c
@@ -359,15 +366,20 @@ nbr_table_add_lladdr(nbr_table_t *table, const linkaddr_t *lladdr, nbr_table_rea

     /* Set link-layer address */
     linkaddr_copy(&key->lladdr, lladdr);
-  }
-
-  /* Get item in the current table */
-  item = item_from_index(table, index);

-  /* Initialize item data and set "used" bit */
-  memset(item, 0, table->item_size);
-  nbr_set_bit(used_map, table, item, 1);
+    /* Get item in the current table */
+    item = item_from_index(table, index);

+    /* Initialize item data... */
+    memset(item, 0, table->item_size);
+    /* and set "used" bit */
+    nbr_set_bit(used_map, table, item, 1);
+  } else {
+    /* Get item in the current table */
+    item = item_from_index(table, index);
+    /* Set "used" bit */
+    nbr_set_bit(used_map, table, item, 1);
+  }
 #if DEBUG
   print_table();
 #endif
```

tscheffl, 2016-12-22:
I had high hopes that your code fixed a segfault that I see in one of my projects (MQTT client run on `minimal-net` platform without RPL).
It starts fine and if I constantly `ping` the client it stays up. 
When I terminate the `ping` and the client posts publish messages in regular intervals it crashes:

```
==9410== Invalid read of size 1
==9410==    at 0x8056B4F: nbr_get_bit (nbr-table.c:134)
==9410==    by 0x8056C9D: nbr_table_next (nbr-table.c:266)
==9410==    by 0x805383F: uip_ds6_neighbor_periodic (uip-ds6-nbr.c:311)
==9410==    by 0x805326A: uip_ds6_periodic (uip-ds6.c:187)
==9410==    by 0x8055D09: eventhandler (tcpip.c:483)
==9410==    by 0x8055D09: process_thread_tcpip_process (tcpip.c:827)
==9410==    by 0x80500A2: call_process (process.c:190)
==9410==    by 0x8050220: do_event (process.c:296)
==9410==    by 0x8050220: process_run (process.c:310)
==9410==    by 0x804F3DD: main (contiki-main.c:348)
==9410==  Address 0x70577d3 is not stack'd, malloc'd or (recently) free'd
==9410== 
==9410== 
==9410== Process terminating with default action of signal 11 (SIGSEGV)
==9410==  Access not within mapped region at address 0x70577D3
```

CogPy, 2017-01-18:
I only run CoAP and no MQTT. I haven't seen a SIGSEGV though, is it dumped
on the UART?

tscheffl, 2017-01-24:
The SIGSEGV is thrown by the Linux Kernel (I am running Contiki via a `minimal-net` build directly in a Linux VM).

When I start the Contiki process from the command-line, it simply crashes with no output on the terminal (although one can use `echo $?` to determine the exit-code). 
So, I then did a fresh compile with debug options enabled(`-g`) and started the Contiki process under Valgrind (`valgrind my_program`).