contiki-os/contiki #1951: contiki 3.0 doesnt burn code on zolertia mote z1

sairarafique, 2016-11-27:
when we do same experiments with 2.7 its works on z1 but contiki 3.0 doesn't work

alignan, 2016-11-27:
We need more information, would you mind at least detailing what you are doing?

sairarafique, 2016-11-27:
specifically i have to find antenna polarization between Z1 motes effects on RSSI, LQI, ETX but generally when ever i try to burn any script on z1 using contiki os it works with contiki 2.7
but when i try to upload and run same scripts using contiki 3.0 it starts giving errors

alignan, 2016-11-27:
Again, this is not helpful at all.

Please provide information that allow anyone to replicate what you are doing and try to provide a solution, i.e work environment (are you using Instant Contiki or other), commit, commands used to flash the device, etc.  Contiki 3.0 has been out for quite a while, and it is unlikely  this has been overlooked until now.

Does the Z1 mote is recognized by your system? what is the output when running the `make motelist` command?

Please document this issue better

closed 2016-12-04 by alignan