contiki-os/contiki #1953: Unable to establish border router connection

Nickkolos, 2016-11-28:
I followed a tutorial on "IPv6 Routing"
Sky motes set as a border router created using  /contiki/examples/ipv6/rpl-borderrouter/border-router.c
1 motes created using /contiki/examples/ipv6/rpl-collect/udp-sink.sky
9 other motes /contiki/examples/ipv6/rpl-collect/udp-sender.sky

The following are typed into the terminal:

user@instant-contiki:~/contiki$ cd tools
user@instant-contiki:~/contiki/tools$ make tunslip6
make: `tunslip6' is up to date. 
user@instant-contiki:~/contiki/tools$ sudo ./tunslip6 -a 127.0.0.1 aaaa::1/64 
[sudo] password for user:  

Output is as follows:

slip connected to ``127.0.0.1:60001''
opened tun device ``/dev/tun0''
ifconfig tun0 inet `hostname` up
ifconfig tun0 add aaaa::1/64
ifconfig tun0 add fe80::0:0:0:1/64
ifconfig tun0

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:127.0.1.1  P-t-P:127.0.1.1  Mask:255.255.255.255
          inet6 addr: fe80::1/64 Scope:Link
          inet6 addr: aaaa::1/64 Scope:Global
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

Rime started with address 0.18.116.1.0.1.1.1
MAC 00:12:74:01:00:01:01:01 Contiki-2.6-2450-geaa8760 started. Node id is set to 1.
nullsec CSMA ContikiMAC, channel check rate 8 Hz, radio channel 26, CCA threshold -45
Tentative link-local IPv6 address fe80:0000:0000:0000:0212:7401:0001:0101
Starting 'Border router process' 'Web server'
*** Address:aaaa::1 => aaaa:0000:0000:0000
Got configuration message of type P
Setting prefix aaaa::
Server IPv6 addresses:
 aaaa::212:7401:1:101
 fe80::212:7401:1:101

However , addesses of motes still remain at fe80 instead of aaaa as expected from the video tutorials found online.
[image]

can anybody please advice on what i am doing wrongly?

davidsantosb, 2016-12-13:
Hey, I am working on this issue too. Every border router is dodag Root, and so is every udp sink. Senders subscribe to just 1 instance, on this example they join the sink. I am trying to find the way to use Border-Router and Udp-sink simultaneously. Still the best idea i got so far is to make udp-sink the border-router.

Hope it helps. I will tell you if i can solve this.

g-oikonomou, 2017-01-10:
https://github.com/contiki-os/contiki/wiki#Help

closed 2017-01-10 by g-oikonomou