contiki-os/contiki #1955: #define RPL_CONF_OF_OCP RPL_OCP_OF0 crash cooja simulator

meetbshah, 2016-11-29:
Hi, 
I am trying to use OF0 routing. 

I have added "#define RPL_CONF_OF_OCP RPL_OCP_OF0" in rpl conf.h and tried to run ipv6/rpl-udp example in cooja. Cooja crashes every time. I can run same example with mrhop config.

pablocorbalan, 2016-11-30:
To use OF0, you can directly add the following to your application `project-conf.h` file:
```
#undef RPL_CONF_OF
#define RPL_CONF_OF rpl_of0
```

This should work. In fact, some examples such as `er-rest-example` are configured to use OF0 that way.

simonduq, 2016-11-30:
Hi,

Not anymore @pablocorbalan :)
The new way to configure OF0 is:
(1) embed support for OF0, by making sure rpl_of0 is in the array RPL_CONF_SUPPORTED_OFS
(2) ask the root to advertise OF0 as the OF in use in its network , setting RPL_CONF_OF_OCP to RPL_OCP_OF0

Now, the fact you got a crash smells like a bug anyway. Nodes should have simply complained that they don't support the OF, at runtime, when receiving a DIO with the OF0 OCP. Worth investigating.

pablocorbalan, 2016-11-30:
It looks like I am getting very rusty with Contiki. I need to catch up with the latest changes :). Sorry for my outdated comment. Then, I guess you just need to add the following to the `project-conf.h`:
```
#define RPL_CONF_SUPPORTED_OFS {&rpl_of0}
#define RPL_CONF_OF_OCP RPL_OCP_OF0
```

@simonduq shall we then remove all the references to `RPL_CONF_OF`? See:
https://github.com/contiki-os/contiki/search?utf8=✓&q=RPL_CONF_OF
I could make a PR if required.

Btw, I confirm that only setting `#define RPL_CONF_OF_OCP RPL_OCP_OF0` makes Cooja crash with the following errors:

### Contiki Error:
```
Writing to read-only P1IN

Stack Trace: number of calls: 5 PC: $0eb50
  memset (memset.c) called from PC: $0982e (elapsed: 43)
  rpl_set_prefix (memset.c) called from PC: $04458 (elapsed: 261)
  process_thread_udp_server_process (local in udp-server.c) called from PC: $061b6 (elapsed: 16033)
  call_process (local in process.c) called from PC: $06346 (elapsed: 16068)
  process_run (memset.c) called from PC: $042b8 (elapsed: 16645)
```

### Java Stack trace:
```
org.contikios.cooja.ContikiError
	at org.contikios.cooja.mspmote.MspMote.execute(MspMote.java:341)
	at org.contikios.cooja.mspmote.MspMote.execute(MspMote.java:298)
	at org.contikios.cooja.motes.AbstractWakeupMote$1.execute(AbstractWakeupMote.java:47)
	at org.contikios.cooja.Simulation.run(Simulation.java:281)
	at java.lang.Thread.run(Thread.java:745)
Caused by: se.sics.mspsim.core.EmulationException: Writing to read-only P1IN
	at se.sics.mspsim.core.IOPort.writePort(IOPort.java:280)
	at se.sics.mspsim.core.IOPort.write(IOPort.java:369)
	at se.sics.mspsim.core.IOSegment.write(IOSegment.java:51)
	at se.sics.mspsim.core.MSP430Core$1.write(MSP430Core.java:171)
	at se.sics.mspsim.core.MSP430Core.emulateOP(MSP430Core.java:2156)
	at se.sics.mspsim.core.MSP430.stepMicros(MSP430.java:256)
	at org.contikios.cooja.mspmote.MspMote.execute(MspMote.java:337)
	... 4 more
```
Setting both values, i.e., `RPL_CONF_SUPPORTED_OFS`and `RPL_CONF_OF_OCP` works fine for me.

simonduq, 2016-11-30:
Yes please PR the config fix!
As for the crash, do you think the problem could be that we don't have a return value for `rpl_join_instance`, and therefore do not abort `rpl_process_dio` when joining the instance fails?