contiki-os/contiki #1959: [Re-mote Zolertia] cc2538dk Cannot autodetect firmware file type: Assuming .bin

zhangtuoparis13, 2016-11-30:
Hi all, 

I have a problem in flashing the cc2538dk,
`make cc2538-demo.upload
using saved target 'cc2538dk'
  CC        ../../cpu/cc2538/./ieee-addr.c
  CC        ../../cpu/cc2538/cc2538.lds
  CC        ../../cpu/cc2538/./startup-gcc.c
  CC        cc2538-demo.c
  LD        cc2538-demo.elf
arm-none-eabi-objcopy -O binary --gap-fill 0xff cc2538-demo.elf cc2538-demo.bin
python ../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v -a 0x00200000 cc2538-demo.bin
Opening port /dev/ttyUSB0, baud 500000
Reading data from cc2538-demo.bin
**Cannot auto-detect firmware filetype: Assuming .bin**
Connecting to target...
CC2538 PG2.0: 512KB Flash, 32KB SRAM, CCFG at 0x0027FFD4
Primary IEEE Address: 00:12:4B:00:06:0D:60:BA
Erasing 524288 bytes starting at address 0x00200000
    Erase done
Writing 524288 bytes starting at address 0x00200000
Write 16 bytes at 0x0027FFF0F8
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xb60ae7cd)
rm cc2538-demo.co obj_cc2538dk/startup-gcc.o
`
Normaly, the

>  Cannot auto-detect firmware filetype: Assuming .bin
should be 

> Firmware file: Raw Binary

but then the output seems correctlly, 
someone can help to figure the issues?
thx a lot!

alignan, 2016-11-30:
Would you mind editing the title? is it the `RE-Mote`  or the `CC2538DK` the issue here? not precisely clear

zhangtuoparis13, 2016-11-30:
@alignan sorry i'm new in using this , i actually have a RE-mote Zolertia . But what is the main difference of RE-Mote  and CC2538DK?

alignan, 2016-11-30:
Both uses the CC2538, but each have different things... try to compile using the right target :smile:

For example the `examples/zolertia/zoul/zoul-demo` application

zhangtuoparis13, 2016-11-30:
@alignan ok, this time 
`~/contiki/examples/zolertia/zoul$ make zoul-demo.upload`
then the output is:
```
using saved target 'zoul'
  CC        ../../../cpu/cc2538/./ieee-addr.c
  CC        ../../../cpu/cc2538/cc2538.lds
  CC        ../../../cpu/cc2538/./startup-gcc.c
  CC        zoul-demo.c
  LD        zoul-demo.elf
arm-none-eabi-objcopy -O binary --gap-fill 0xff zoul-demo.elf zoul-demo.bin
Flashing /dev/ttyUSB0
Opening port /dev/ttyUSB0, baud 500000
Reading data from zoul-demo.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
CC2538 PG2.0: 512KB Flash, 32KB SRAM, CCFG at 0x0027FFD4
Primary IEEE Address: 00:12:4B:00:06:0D:60:BA
Erasing 524288 bytes starting at address 0x00200000
    Erase done
Writing 516096 bytes starting at address 0x00202000
Write 8 bytes at 0x0027FFF8F00
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0x45381a05)
rm obj_zoul/startup-gcc.o zoul-demo.co
```

The proeblem **Cannot auto-detect firmware filetype: Assuming .bin**  persist also.

But it seems work when i `make login`

```
~/contiki/examples/zolertia/zoul$ make login
using saved target 'zoul'
../../../tools/sky/serialdump-linux -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0 (115200) [OK]
-----------------------------------------
Counter = 0x00000008
VDD = 3283 mV
Temperature = 35952 mC
ADC1 = 2316 raw
ADC3 = 288 raw
-----------------------------------------
Counter = 0x00000009
VDD = 3283 mV
Temperature = 35714 mC
ADC1 = 2312 raw
ADC3 = 288 raw
```
i am really confused...

alignan, 2016-11-30:
The compiler will produce `elf`, `bin` and `hex` if specified, but the flash tool assumes `bin`, and then flash the device (as shown in the log).

What are the expected behaviour you are currently missing? is this an inconvenient?

g-oikonomou, 2016-11-30:
Do you have python-magic installed?

alignan, 2016-11-30:
More instructions [here](https://github.com/Zolertia/Resources/wiki/Program-the-Zolertia-platforms#zolertia-re-mote-platform)

g-oikonomou, 2016-11-30:
> More instructions here

Also here: https://github.com/JelmerT/cc2538-bsl#dependencies

zhangtuoparis13, 2016-11-30:
@alignan @g-oikonomou Yes, it works with zoul-demo this time, i confirm that i have missed the `python-magic`. 
Another question, why can't I use the cc2538-demo?

alignan, 2016-11-30:
Yes, you can as well (in the first log you have already compiled and ran the example).

closed 2016-11-30 by alignan