contiki-os/contiki #1968: section `.data' will not fit in region `rom'

lawpeter, 2016-12-03:
Hello, 
I try to run an example (ipv6/rpl-tsch/) and I got this error. 
Even if I downloaded a new contiki Instant virtual machine, it is still error. 

/usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/bin/ld: **node.z1 section `.data' will not fit in region `rom'**
/usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/bin/ld: **region `rom' overflowed by 2 bytes**

Thank you.

lawpeter, 2016-12-03:
I think there are some things wrong with RPL or TSCH, because I changed the new computer or contiki instant, but it is still same error.

alignan, 2016-12-03:
I tested using `msp430-gcc-4.7.2` as it is working OK, this is the same version the Travis tests use.  You could either upgrade your toolchain version, or reduce some fat in the code.

lawpeter, 2016-12-04:
Thank you very much, 
That's exactly what I need, it's working well now :+1:

closed 2016-12-04 by lawpeter

yadeleke, 2017-02-14:
Hi,

Could you please tell me the steps you took to solve this problem. Thanks

alignan, 2017-02-14:
Have you at least tried the above steps? hard to know what you have tried already...

yadeleke, 2017-02-14:
I have fixed it now. My issue was i didn't know how to upgrade toolchain.

rhr407, 2017-05-17:
@alignan Can I ask you how did you upgrade from msp430-gcc 4.7.0 to 4.7.2?

rhr407, 2017-05-17:
@lawpeter  Can I ask you how did you upgrade from msp430-gcc 4.7.0 to 4.7.2?

lawpeter, 2017-05-18:
Hi, 
This instruction work perfectly for me. Hope it's useful.  
https://github.com/pksec/msp430-gcc-4.7.3

rhr407, 2017-05-25:
Hello there,
 I am also working on z1 and came around with the 2 bytes error as mentioned by  @lamnd09. I have figured out one mistake in sections/segments file memory.x found in /opt/mspgccx/msp430/lib/ldscripts/msp430f2617 folder. Just go to the following line 

rom (rx)         : ORIGIN = 0x3100, LENGTH = 0xcebe /* END=0xffc0, size 52926 */

and replace with 

rom (rx)         : ORIGIN = 0x3100, LENGTH = 0xcec0 /* END=0xffc0, size 52926 */

Cheers...

atiselsts, 2017-05-25:
Interesting catch. But I wonder if there's some specific reason for the length it to be exactly 52926 in the first place? I.e. this "fix" might actually break other things.

Besides, the `0xcebe` length value remains even in `msp430-4.7.3`, and my impression was that this "overflow by 2" bug does not appear in that version.

rhr407, 2017-06-01:
For your first question, I would suggest referring to msp430 data sheet memory map section where there is a vectors section in the middle of rom section. The weired thing is that I have found not a single way to to let msp430-gcc automatically allocate the extra .text section to the far_rom which should be the logical choice for memory allocation after the rom section is depleted. 
Unfortunately, I am facing this error till now even after doing the above mentioned modification. I am quite sure this is a bug even in 4.7.3 and it is indeterministic too. For example, many a times it runs with no error but sometimes it is hard to believe the same code compilation gives this very error.This bug  need to be removed in msp430-gcc next release.

knaggita, 2017-06-03:
It has been puzzling me for a while now

mbaddeley, 2017-11-16:
Just to add a little bit to this, I tried the 'fix' @rhr407 suggested (editing memory.x) for the exp5438, however the values are all correct in that, so although that may have been a bug for the msp430f2617, it isn't for the msp430f5438, and the issue persists.

vishal623, 2018-07-23:
Hello,

I followed instruction as per given by this link: https://github.com/pksec/msp430-gcc-4.7.3

After these steps when I checked my msp430-GCC version, it shows me 4.6.3 only.

My msp430 version is not updated. I want to simulate er-rest example on a Z1 platform.

Any solution???

rhr407, 2018-07-23:
you have to add that path in .bashrc file in home folder

$sudo gedit .bashrc

rhr407, 2018-07-23:
please also remove existing 4.6 version

$sudo apt-get remove gcc-msp430

vishal623, 2018-07-23:
I already added that path in bashrc. This is the statement that I added.

export PATH=$PATH:/opt/compilers/mspgcc-4.7.3/bin

vishal623, 2018-07-23:
Now, It shows me 4.7.3. 

But when I run the simulation and try to burn the code on Z1 mote it shows me 

msp430-gcc: Command not found.

rhr407, 2018-07-23:
did u extracted the downloaded  folder (mspgcc-4.7.3) exactly in
/opt/compilers?

rhr407, 2018-07-23:
restart your system. it will load .bashrc only then

vishal623, 2018-07-23:
Ok.let me do it

vishal623, 2018-07-23:
Sir, it is giving me the same error. 

This is the path for msp430

/opt/compilers/mspgcc-4.7.3

Inside, there are 6 folders and one of them is msp430

I have added this path in /etc/environment

/opt/compilers/mspgcc-4.7.3/msp430/bin

and in bashrc 

export PATH=$PATH:/opt/compilers/mspgcc-4.7.3/bin

rhr407, 2018-07-23:
it's not bashrc file in which you have to paste that link. it's .bashrc

vishal623, 2018-07-23:
Sir, 

I have pasted in ~/.bashrc file. In the same file I have set all the path.

rhr407, 2018-07-23:
Please do the same for .profile file and then log out and log in your PC again.

greg-king5, 2018-07-23:
Don't put any PATH lines into `~/.bashrc`!  That file is for things that aren't exported.  Put all PATH lines into `~/.profile` (or `~/.bash_profile` if it exists).

Also, I think that the names of the MSP430 tools changed.  I don't remember the details; but, I think that the new names have the version number in them.  You must edit "cpu/msp430/Makefile.msp430".  Fix the spellings of the `"msp430-*"` tool-names.

rhr407, 2018-07-24:
4.7.3 work fine for me.

vishal623, 2018-07-24:
I have added this path in the ~/.profile file.  export
PATH=$PATH:/opt/compilers/mspgcc-4.7.3/bin

But there is no change in the output. Same error.

Then after I changed in this lines:

CC       = msp430-gcc  to msp430-gcc-4.7.3 or msp430-4.7.3
LD       = msp430-gcc  to msp430-gcc-4.7.3 or msp430-4.7.3 or msp430-ld

Whatever I wrote, it shows me command not found.

vishal623, 2018-07-24:
I checked my gcc and g++ version. It is 4.7.4.

rhr407, 2018-07-24:
@greg-king5 : You are right it's .profile file not .bashrc. I was mistaken. sorry for that. @vishal623 : Have you logged in your PC again. It must work.

vishal623, 2018-07-24:
Yes, sir.

I have followed the steps that were given by
https://github.com/pksec/msp430-gcc-4.7.3.

   1. I didn't remove the olde version of msp430.
   2. I have extracted the msp430-gcc-4.7.3 to /opt/compilers/. Inside that
   there will be one more zip file, so I also extracted it and mv outside of
   that directory and gave the name of msp430-gcc-4.7.3.
   3. I have added the path in /etc/environment. which is
   msp430-gcc-4.7.3/msp430/bin
   4. I have added the path in ~/.profile. which is msp430-gcc-4.7.3/bin
   5. I have also updated the CC and LD variables in Makefile.msp430

When I checked the version of msp430, it shows me 4.7.3, but when I run the
border router on sky mote, it shows ms msp430-gcc-4.7.3 command not found.
The reason behind it that It has a small amount of memory.
Even when I run the er-rest-client on z1, it shows me the same error.

For any other mote, the error will be the same.

rhr407, 2018-07-24:
Can you send me a screen shot of contents of the foler /opt/compilers/msp430-gcc-4.7.3 please?

rhr407, 2018-07-24:
[image]
It should be like this. 
[image]
an the bin folder must be like this.

vishal623, 2018-07-24:
[image: msp430-gcc-4.7.3.png]
main directory

[image: msp430-gcc-4.7.3-bin.png]
​bin folder
​

vishal623, 2018-07-24:
sure. 

/opt/compilers/msp430-gcc-4.7.3
[image]

/opt/compilers/msp430-gcc-4.7.3/bin

[image]

rhr407, 2018-07-24:
It seems fine. I am surprised why are you still facing the issue.

rhr407, 2018-07-24:
Please change the folder name from msp430-gcc-4.7.3 to msp430gcc-4.7.3
I hope this is the error.

vishal623, 2018-07-24:
Should I change in Makefile.msp430?

rhr407, 2018-07-24:
no

rhr407, 2018-07-24:
yes if you have changed it previously. There is no need to change the make file. Revert it to its default state as it was already.

vishal623, 2018-07-25:
Now, I am able to compile the code of the border router on sky mote. But I
am not able to compile the er-rest-client code on Z1 mote. For that I have
to optimize the code.

Thanks for helping me.

rhr407, 2018-07-25:
what is the error?

vishal623, 2018-07-25:
"ROM overflow by **** bytes".

vishal623, 2018-07-25:
I am able to compile er-example-client code on Z1 mote but when I compiled
server code on Z1 it shows me the following error.

/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: er-example-server2.z1
section `.rodata' will not fit in region `rom'
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: section .vectors
loaded at [000000000000ffc0,000000000000ffff] overlaps section .rodata
loaded at [000000000000fa38,000000000001005d]
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: region `rom'
overflowed by 584 bytes
/usr/lib/gcc/msp430/4.6.3/mcpu-430x/mmpy-16/libcrt0.a(_copy_data.o): In
function `__do_copy_data':
/build/buildd/gcc-msp430-4.6.3~mspgcc-20120406/./gcc-4.6.3/gcc/config/msp430/crt0.S:195:
relocation truncated to fit: R_MSP430_16_BYTE against symbol
`__data_load_start' defined in *ABS* section in er-example-server2.z1
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mcpu-430x/mmpy-16/libc.a(vuprintf.o):
In function `vuprintf':
/build/buildd/msp430-libc-20120224/src/./stdlib/vuprintf.c:433: relocation
truncated to fit: R_MSP430_16_BYTE against `no symbol'
collect2: ld returned 1 exit status
../../Makefile.include:280: recipe for target 'er-example-server2.z1' failed
make: *** [er-example-server2.z1] Error 1
rm obj_z1/res-battery.o obj_z1/res-leds.o obj_z1/res-mirror.o
obj_z1/res-toggle.o obj_z1/res-radio.o obj_z1/res-b1-sep-b2.o
er-example-server2.co obj_z1/contiki-z1-main.o obj_z1/res-push.o
obj_z1/res-separate.o obj_z1/res-chunks.o obj_z1/res-light.o
obj_z1/res-sht11.o obj_z1/res-sub.o obj_z1/res-hello.o obj_z1/res-event.o
Process returned error code 2

I already updated the msp430 version still in this error it shows me 4.6.3.

rhr407, 2018-07-25:
keep printfs minimum

vishal623, 2018-07-25:
I am using the default code of er-rest-client and er-rest-server which is
given in er-rest-example folder.

rhr407, 2018-07-25:
set the DEBUG 0 in er-coap.c file or wherever you think necessary so that
no debug strings get printed on terminal.

vishal623, 2018-07-30:
Hello sir,

I set the DEBUG 0 or DEBUG_NONE in er-coap.c as well as in other files. Why it shows me the error in msp430gcc-4.6.3. I already set the msp430gcc-4.7.3.

rhr407, 2018-07-30:
Uninstall 4.6

vishal623, 2018-07-31:
When I uninstalled the gcc-msp430 it shows me "msp430-gcc: command not
found".

When I checked the version of msp430 by using command
"msp430-gcc --version", it shows me 4.7.3.

vishal623, 2018-07-31:
Should I use below command?

sudo apt-get purge --auto-remove gcc-msp430

It will delete configuration and/or data files of gcc-msp430 and it's
dependencies from Ubuntu Trusty. Is it necessary?

One more is

sudo apt-get remove --auto-remove gcc-msp430

It will remove the gcc-msp430 package and any other dependent package which
are no longer needed.

rhr407, 2018-07-31:
no need to use thse commands once 4.6 is uninstalled. please read the
precious posts carefully for 4.7.3

vishal623, 2018-07-31:
Sir, I followed all the steps properly. Here, I am attaching screenshots of
all the steps. PFA.

vishal623, 2018-08-01:
Now, I am also facing problem to compile rpl border router on sky mote.

rhr407, 2018-08-01:
please check if you have the same path /opt/compilers... as mentioned in my
earlier posts.

vishal623, 2018-08-01:
In the post, they mentioned path /opt/compilers/mspgcc-4.7.3

So, I changed the path and recompiled it, but the error is the same
msp430-gcc: command not found

In contiki/cpu/msp430/Makefile.msp430, the CC and LD is msp430-gcc. I also
checked by changing it to msp430-gcc-4.7.3 or mspgcc-4.7.3

But it didn't work.

I tried a lot, but all things went wrong for me. If possible then Could you
please access my system and check once.

Thanks
Vishal

rhr407, 2018-08-01:
yes, I was also going to suggest that. Please email me your skype.

vishal623, 2018-08-01:
Ok sir.

First I have to install it. Give me some time.

vishal623, 2018-08-01:
my skype account name is rathodvishalji.

[15 more replies]