contiki-os/contiki #1974: Tunslip6 broken ?

MartinNPN, 2016-12-05:
Hello,

i actually testing tunslip6 at os x and found out that it didn't work. On a Linux VM it is working as expected but with OS X 10.12.1 it doesn't. 
First i fixed the compiler warning: 
tunslip6.c:135:3: warning: implicit declaration of function 'gettimeofday' is invalid in C99 [-Wimplicit-function-declaration]
by adding the correct header: 
#include <sys/time.h>

Maybe this could be corrected by someone.

But after that i startet with:
sudo ./tunslip6 -B 38400 -s /dev/tty.usbmodemXXXX fd00::1/64
and got:
********SLIP started on ``/dev/tty.usbmodemXXXX''
tunslip6: main: open /dev/tun: Is a directory

Is that a common problem ?
I can use open vpn under os x which although creates a tun device without any problems.

Has anyone an idea ?

Regards,
Martin

greg-king5, 2016-12-06:
See issue #1560.

MartinNPN, 2016-12-06:
Hello,

thanks, that solved the problem, i thought it is so old that it would be fixed already.

closed 2016-12-06 by MartinNPN