contiki-os/contiki #1975: Problem rediscovering CC1310 leaf node with multihop communication

BaN-Co, 2016-12-05:
Hello,

I have a problem with leaf node  with multihop communication.
It seems like after the lifetime of the leaf node will reach zero (for example, if I ping it, while it sleeps), it'll be deleted on BR and it's not added afterwards for a long time.
If the leaf node is connected to the BR directly, then there is no problem.
The problem arise, when I connect the leaf node to the BR via additional router node. On power up everything works fine and all is fine so long as I only communicate with the leaf node, when it's mac is on. But once I try to communicate with the leaf node while it sleeps and the leaf node lifetime will reach 0, then the node is removed from the router node routing table and from the BR. And after this the leaf node keep sending DIOs periodically, but it can't connect again to the network up to a few hours.
If I power cycle the router node, then BR will discover the leaf node in no time again, when it's mac will be on.
If I restart the 6lbr service on BR, then BR will discover the leaf node in no time again, when it's mac will be on.

My configuration:
- My PC (bbbb::c077:dc44:46de:8ca1) - I ping the leaf node from it
- Boarder router (fd00::212:4b00:89ab:cdef) - beaglebone black with CC1310, connected via UART and flashed with slip-radio, running 6lbr.
- Router node (fd00::fd11:2233:4455:6677) - CC1310, not in a leaf mode, the project code is just this:
```
PROCESS(router_demo_process, "CC13xx/CC26xx router process");
AUTOSTART_PROCESSES(&router_demo_process);
PROCESS_THREAD(router_demo_process, ev, data)
{
  PROCESS_BEGIN();

  printf("Router Demo Process\n");

  while(1) {
    PROCESS_WAIT_EVENT();
  }

  PROCESS_END();
}
```
- Leaf node (fd00::211:2233:4455:6677) - CC1310 in leaf mode, based on very-sleepy-demo, wakes every 60 seconds and keep mac on for 10 seconds.

The contiki and 6lbr code is up to date.

I'm new to contiki and I don't understand how everything should work quite yet, so I made a few logs and maybe someone can point out some tips on how to debug this problem.

Here is the router node log, from the start up, discover, ping the leaf node, reaching 0 lifetime, removing routes and can't discover leaf node at the end:
[Leaf node.zip](https://github.com/contiki-os/contiki/files/631030/Leaf.node.zip)
I can see it periodically sends DIOs, but there's no ACK from leaf node. Also it receives the DIOs from the leaf node.

Here is the leaf node log, after it got removed from the routing tables:
[Router node from the start.zip](https://github.com/contiki-os/contiki/files/631029/Router.node.from.the.start.zip)
I can see it periodically sends DIOs. And also, there is a few times, when it receives the DIO packet from the router node.

Here is the router node log, after the leaf node got removed from the routing tables and the moment, when it got discovered again:
[Router node, when it rediscovered the leaf node.zip](https://github.com/contiki-os/contiki/files/631028/Router.node.when.it.rediscovered.the.leaf.node.zip)

BaN-Co, 2016-12-07:
It's not just a leaf node with multihop problem, it's just more distinctly for them.
I've tested the same situation with the non-leaf node, that was connected directly to the BR without multihop. I've moved the non-leaf mode out of range of the BR and then pinged it, utill it got deleted, then I've move the non-leaf node in the BR range again and it got rediscovered only after the random interval up to 7 minutes minutes later.
Seems like there are some differences in the discovery process of the newly added nodes and the nodes, that got deleted previously due to lifetime zeroing.
The newly added nodes are discovered in a few seconds after power up.
I'll try test some more.

MartinNPN, 2016-12-07:
Hello,

i had discovered the same problem, but not every time. I found out that it could be a problem with the contikimac rdc, because when i use the nullrdc the problem is gone. 
Maybe it is the same as #1970

g-oikonomou, 2016-12-19:
Closing as duplicate of #1878

closed 2016-12-19 by g-oikonomou