contiki-os/contiki #1977: [erbium] Mixed payload when resource value change pending block2 exchange.

sbernard31, 2016-12-05:
I looked at the block2 implementation of erbium.
And it seems that  [`service_cbk`](https://github.com/contiki-os/contiki/blob/master/apps/er-coap/er-coap-engine.c#L137) is called for each `block2 request`. 

But if the value changes between 2 request of the same block2 exchange the final payload built at client side will be a mix of those values. (before and after the value change)

Am I right ? or did I missed something ?

joakimeriksson, 2016-12-15:
Yes, I assume this is the case. And it is likely not much to do about it since a device typically have not enough RAM to handle large block requests. If there is a possibility to do that it is up to the resource to ensure consistency of the values. I guess in a future version it might be possible to allocate a very large CoAP buffer and put all the data there and allow erbium to read out from that.

sbernard31, 2016-12-16:
Thx  @joakimeriksson for the clarification.