contiki-os/contiki #1979: Board Router can't rediscover CC1310 nodes with multihop communication

BaN-Co, 2016-12-07:
Hello,

It seems like after the lifetime of the previously connected nodes will reach zero (for example, if I ping them, while they are out of BR range), they'll be deleted on BR and can't be discovered by BR afterwards.
If the nodes are connected to the BR directly, then there is no problem.
The problem arise, when I connect the end node to the BR via additional router node (BR <-> Router node <-> End node). On power up everything works fine and all is fine. But once I try to communicate with the nodes while they are out of the BR range and the nodes lifetime will reach 0, then the nodes are removed from the BR. And after this the nodes can't connect again to the network.

My configuration:

Boarder router (fd00::212:4b00:89ab:cdef) - beaglebone black with CC1310, connected via UART and flashed with slip-radio, running 6lbr.
Router node (fd00::fd11:2233:4455:6677) - CC1310, the project code is just this:
```
PROCESS(router_demo_process, "CC13xx/CC26xx router process");
AUTOSTART_PROCESSES(&router_demo_process);
PROCESS_THREAD(router_demo_process, ev, data)
{
  PROCESS_BEGIN();

  printf("Router Demo Process\n");

  while(1) {
    PROCESS_WAIT_EVENT();
  }

  PROCESS_END();
}
```
End node (fd00::211:2233:4455:6677) - CC1310, project contains rest resource.

The contiki and 6lbr code is up to date.

Here is how I tested it:

1. Power down nodes.
2. Restart 6lbr.
3. Power up nodes. The router node is in the BR range, end node is in the router node range and out of BR range.
4. Wait until the nodes will be added in the BR.
5. Move the router and end nodes out of BR range.
6. Ping both router and end nodes, until they will be deleted from the BR.
7. Cancel pinging the nodes.
8. Move the router node in the BR range again (end node is in the router node range).
9. The router node and end node can't reconnect to the BR.

Here is some logs:

- Here is the router node log after it got moved out of BR range:
[Router node out of BR range.zip](https://github.com/contiki-os/contiki/files/636330/Router.node.out.of.BR.range.zip)
- Here is the router node log after it got moved back in the BR range:
[Router node in the BR range again.zip](https://github.com/contiki-os/contiki/files/636331/Router.node.in.the.BR.range.again.zip)

These repeating messages can be seen in the log after the router node got back in the BR range:
```
RPL: probing 239  last tx 2 min ago
RPL: Sending prefix info in DIO for fd00::
RPL: Sending unicast-DIO with rank 65535 to fe80::212:4b00:89ab:cdef
Upper layer checksum len: 76 from: 40
sicslowpan output: sending packet len 116
before compression (76): 60000000004c3a40fe80000000000000fd11223344556677fe8000000000000002124b0089abcdef9b014a461e19ffff900a0000fd0000000000000002124b0089abcdef040e00080c0a070001000001001e0100081e4040000000000000000000000000fd000000000000000000000000000000
sicslowpan output: header of len 3
RPL: MOP 2 OCP 1 rank 65535 dioint 20, nbr count 2
RPL: nbr 239   256,  1216 =>  1472 -- 16 f  (last tx 2 min ago)
RPL: nbr 119 65535,   160 => 65535 -- 10 f  (last tx 1 min ago)
RPL: end of list
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
mac_callback_t #0x363d ptr #0x0 status 2 num_tx 8
mac: noack after 8 tx
RPL: rpl_link_neighbor_callback triggering update
RPL: rpl_process_parent_event recalculate_ranks
RPL: DIO Timer triggered
RPL: Sending prefix info in DIO for fd00::
RPL: Sending a multicast-DIO with rank 65535
Upper layer checksum len: 76 from: 40
sicslowpan output: sending packet len 116
before compression (76): 60000000004c3a40fe80000000000000fd11223344556677ff02000000000000000000000000001a9b01ee571e19ffff900a0000fd0000000000000002124b0089abcdef040e00080c0a070001000001001e0100081e4040000000000000000000000000fd000000000000000000000000000000
sicslowpan output: header of len 4
RPL: Scheduling DIO timer 49506 ticks in future (sent)
RPL: MOP 2 OCP 1 rank 65535 dioint 20, nbr count 2
RPL: nbr 239   256,  1225 =>  1481 -- 16 f  (last tx 1 min ago)
RPL: nbr 119 65535,   160 => 65535 -- 10 f  (last tx 2 min ago)
RPL: end of list
contikimac: send broadcast
contikimac: send (strobes=7, len=95, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 0 num_tx 1
mac: sent after 1 tx
mac_callback_t #0x363d ptr #0x0 status 0 num_tx 1
mac: sent after 1 tx
contikimac: data (79)
sicslowpan input: IPHC
IPHC: next header inline: 58
Uncompressing 2 + 0 => fe80::211:2233:4455:6677
Uncompressing 2 + 0 => fe80::fd11:2233:4455:6677
sicslowpan input: IP packet ready (length 116)
after decompression 76:60000000004c3a40fe800000000000000211223344556677fe80000000000000fd112233445566779b011feb1e19ffff90020000fd0000000000000002124b0089abcdef040e00080c0a070001000001001e0100081e4040000000000000000000000000fd000000000000000000000000000000
IPv6 packet received from fe80::211:2233:4455:6677 to fe80::fd11:2233:4455:6677
icmp6_input: length 116 type: 155
Upper layer checksum len: 76 from: 40
RPL: Received a DIO from fe80::211:2233:4455:6677
RPL: Incoming DIO (id, ver, rank) = (30,25,65535)
RPL: Incoming DIO (dag_id, pref) = (fd00::212:4b00:89ab:cdef, 0)
RPL: DIO option 4, length: 14
RPL: DAG conf:dbl=8, min=12 red=10 maxinc=1792 mininc=256 ocp=1 d_l=30 l_u=256
RPL: DIO option 8, length: 30
RPL: Copying prefix information
RPL: Prefix announced in DIO
RPL: Prefix set - will announce this in DIOs
rpl_set_prefix - prefix NON-NULL
Set dag fd00::212:4b00:89ab:cdef lifetime to 1048
RPL: Received consistent DIO
RPL: preferred DAG fd00::212:4b00:89ab:cdef, rank 65535, min_rank 512, parent rank 65535, link metric 160
RPL: Unacceptable rank 65535 (Current min 512, MaxRankInc 1792)
RPL: Nullifying parent fe80::211:2233:4455:6677
RPL: The candidate parent is rejected
RPL: rpl_process_parent_event recalculate_ranks
RPL: Unacceptable rank 65535 (Current min 512, MaxRankInc 1792)
RPL: Nullifying parent fe80::211:2233:4455:6677
RPL: A parent was dropped

```
Seems like it doesn't receive any ACK from the BR after it sends unicast-DIO to it.
Maybe it somehow related to these issues:
https://sourceforge.net/p/contiki/mailman/contiki-developers/thread/527BB8B2.5040505%40cedint.upm.es/#msg31611350
https://sourceforge.net/p/contiki/mailman/message/30276424/

BaN-Co, 2016-12-07:
I've made the hardware reset for the slip-radio and restarted 6lbr of the BR. In 27 minutes after this, the nodes got rediscovered by the BR. I don't know, whatever this was the result of the restart or not, but here is the logs of this process:
- The 6lbr log after the restart:
[6lbr_2.zip](https://github.com/contiki-os/contiki/files/636713/6lbr_2.zip)
- The router node after the BR restart:
[Router node after the BR restart.zip](https://github.com/contiki-os/contiki/files/636718/Router.node.after.the.BR.restart.zip)

BaN-Co, 2016-12-07:
Seems like it's the slip-radio or 6lbr problem.
I've reproduced this problem once again and in the state, in which router node is in the BR range and still can't be discovered, I've added new router node, that wasn't previously in the network in the BR range. The old router node saw the DIO from this new router node (I've started the new router near the old one, then moved it near BR after a few seconds), but there is no info about this node in the 6lbr log (trace level) and the BR can't discover this new node.
I've restarted the 6lbr service and there wasn't any immediate improvement.
I've power cycled the new router node two times - to no avail.
I've waited for 40 minutes and then I've stopped 6lbr, made hardware reset of the slip-radio and once again started 6lbr.
After this, the nodes weren't discovered in 2 minutes, so I power cycled the new router node and it got discovered immediately. After this in 1.5 minutes the old router node and end node got discovered too.
I think it's most likely a problem of the slip-radio.

BaN-Co, 2016-12-07:
Tried one more test.
Here is the network configuration:
          BR
        /      \
Node3   Node1
                   |
              Node2
Node3 and Node1 can't see each other, but can see BR.

Here is the test:
1. Moved the Node1 out of BR range.
2. Pinged Node1 and Node2 untill they got deleted in BR.
3. Moved the Node1 back in the BR range.
4. Waited for 20 minutes, Node1 and Node2 weren't rediscovered.
5. Try to ping the Node3, but it's unreachable, ping until it got deleted from the BR.
After I started pinging the Node3, there were a lot of "contikimac: data not for us" messages in the Node1 log.
6. I've power cycled the the Node3 - no reaction in the 6lbr log. I've waited for 5 minutes, the Node3 wasn't rediscovered.
7. Restarting the 6lbr. No nodes rediscovered after 5 minutes.
8. Power off the Node3.
9. Restarting the 6lbr, wait for it to get ready.
10. Power on the Node3.
11. There is no Node3 activity in the 6lbr log.
12. Stopping the 6lbr.
13. Hardware reset of the slip-radio.
14. Starting the 6lbr.
15. The Node3 was discovered in a few seconds. The Node1 and Node2 were discovered after 4 minutes.

BaN-Co, 2016-12-08:
When I've tried to debug the problem with rediscovering nodes by the BR, I've stumble upon a new problem, maybe related to the main one:

The network configuration:

BR <-> Node1 <-> Node2

Node2 can't see BR.

BR - fd00::212:4b00:89ab:cdef
Node1 - fd00::211:2233:4455:6677
Node1 - fd00::2ff:2233:4455:6677

1. All nodes are discovered by BR and can be reached.
2. Move the Node1 out of the BR range.
3. Ping the Node1 and Node2 untill they are deleted from BR.
4. Move the Node1 back in the BR range.
5. Node1 was rediscovered in a minute, but Node2 can't be discovered by the BR.

Here is the Node1 log after the 5 step:
[Node1 log.zip](https://github.com/contiki-os/contiki/files/640207/Node1.log.zip)
Here is the Node2 log after the 5 step:
[Node2 log.zip](https://github.com/contiki-os/contiki/files/640206/Node2.log.zip)

There is a problem, that Node1 can't send the ACK to the Node2 in response to the Node2 sending unicast-DIO and there is a lot "contikimac: Drop duplicate" in the Node1 and Node2 logs.
Also, there are these messages in the Node1 log, maybe it's somehow related:
```
RPL: Unacceptable rank 65535 (Current min 512, MaxRankInc 1792)
RPL: Nullifying parent fe80::211:2233:4455:6677
```

BaN-Co, 2016-12-09:
Regarding the main problem, when the problem occures, the slip-radio doesn't send ACKs to the nodes unicast-DIOs to it.
I'm trying to to print the slip-radio debug information, but there are some problems with it on which I'm working  right now.

BaN-Co, 2016-12-09:
I've enabled the debug output for the slip-radio, and here is the logs of the situation, where BR can't rediscover previously deleted nodes:

- In 6lbr log I can see the slip-radio sending the broadcast packets, but it doesn't receive anything:
```
2016-12-09 17:35:18.078216: PACKET: PF: bridge_output: Sending packet to Broadcast
2016-12-09 17:35:18.089488: PACKET: PF: wireless_output: sending packet
2016-12-09 17:35:18.095232: PACKET: BR-RDC: write: 80 (sid: 9, cb: 1)
2016-12-09 17:35:18.102084: DUMP: BR-RDC: 
         7a3b3a1a 9b011285 1e3e0100 901c0000 
         fd000000 00000000 02124b00 89abcdef 
         040e0008 0c0a0700 01000001 001e0100 
         081e4040 00000000 00000000 00000000 
         fd000000 00000000 00000000 00000000 
         
2016-12-09 17:35:18.120387: PACKET: SLIP: write: 108
2016-12-09 17:35:18.125338: DUMP: SLIP: 
         21530903 0a86640b 00010e00 0141d864 
         cdabffff efcdab89 004b1200 7a3b3a1a 
         9b011285 1e3e0100 901c0000 fd000000 
         00000000 02124b00 89abcdef 040e0008 
         0c0a0700 01000001 001e0100 081e4040 
         00000000 00000000 00000000 fd000000 
         00000000 00000000 00000000 
2016-12-09 17:35:18.359736: PACKET: SLIP: read: 521
2016-12-09 17:35:18.370779: DUMP: SLIP: 
         53522d53 494e3a20 31303820 27215327 
         0a736c69 702d7261 64696f3a 2073656e 
         64696e67 20392028 39352062 79746573 
         290a3231 35333039 30333061 38363634 
         30623030 30313065 30303031 34316438 
         36346364 61626666 66666566 63646162 
         38393030 34623132 30303761 33623361 
         31613962 30313132 38353165 33653031 
         30303930 31633030 30306664 30303030 
         30303030 30303030 30303032 31323462 
         30303839 61626364 65663034 30653030 
         30383063 30613037 30303031 30303030 
         30313030 31653031 30303038 31653430 
         34303030 30303030 30303030 30303030 
         30303030 30303030 30306664 30303030 
         0a31352e 342d494e 3a202031 20303031 
         323a3462 30303a38 3961623a 63646566 
         20203030 30303a30 3030303a 30303030 
         3a303030 30203935 20283935 290a636f 
         6e74696b 696d6163 3a207365 6e642062 
         726f6164 63617374 0a636f6e 74696b69 
         6d61633a 2073656e 64202873 74726f62 
         65733d37 2c206c65 6e3d3935 2c206e6f 
         2061636b 2c206e6f 20636f6c 6c697369 
         6f6e292c 20646f6e 650a6d61 635f6361 
         6c6c6261 636b5f74 20233078 31326420 
         70747220 23307832 30303031 66303020 
         73746174 75732030 206e756d 5f747820 
         310a6d61 633a2073 656e7420 61667465 
         72203120 74780a53 6c69702d 72616469 
         6f3a2070 61636b65 74207365 6e742120 
         7369643a 20392c20 73746174 75733a20 
         302c2074 783a2031 0a
2016-12-09 17:35:18.464008: INFO: SLIP: SR-SIN: 108 '!S'
slip-radio: sending 9 (95 bytes)
15.4-IN:  1 0012:4b00:89ab:cdef  0000:0000:0000:0000 95 (95)
contikimac: send broadcast
contikimac: send (strobes=7, len=95, no ack, no collision), done
mac_callback_t #0x12d ptr #0x20001f00 status 0 num_tx 1
mac: sent after 1 tx
Slip-radio: packet sent! sid: 9, status: 0, tx: 1
2016-12-09 17:35:18.464206: PACKET: SLIP: read: 5
2016-12-09 17:35:18.464371: DUMP: SLIP: 
         21520900 01
2016-12-09 17:35:18.464609: TRACE: SCMD: Got configuration message of type R
[... 149 lines cut ...]
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
contikimac: send unicast to 0012:4b00:89ab:cdef
contikimac: send (strobes=7, len=100, no ack, no collision), done
mac_callback_t #0xb111 ptr #0x20001ba8 status 2 num_tx 1
mac: noack after 1 tx
mac_callback_t #0x363d ptr #0x0 status 2 num_tx 8
mac: noack after 8 tx
RPL: rpl_link_neighbor_callback triggering update
RPL: rpl_process_parent_event recalculate_ranks
```

BaN-Co, 2016-12-09:
I've added the debug info print in the cc26xx_rf_cpe0_isr() function in the ./cpu/cc26xx-cc13xx/rf-core.c and when the problem happens and the BR can't rediscover the nodes, I can see that in the slip-radio the interrupts on the packet send (when sending broadcast) are triggered, but there are no interrupts on the packet receive. In the Node1 log I can see the interrupts, that occurs when it receive the broadcast packet from the BR, when the Node1 sends the ACK and also when Node1 sends the unicast-DIOs to the BR.

g-oikonomou, 2016-12-09:
This sounds like it might be related to #1878? Try using a sniffer and see where/how traffic is getting lost. My guess is that the receiver is not sending ACK frames.

BaN-Co, 2016-12-09:
> My guess is that the receiver is not sending ACK frames.
Are you talking about the node, that got the broadcast message from slip-radio and it doesn't send ACKs?
The configuration is: BR <-> Node1 <-> Node2
When the problem occurs, the Node1 and Node2 continue to communicate normally, only BR can't receive anything. Even if I power up the new Node3 in this state in the range of the BR, the BR can't see this new node messages.
It's more like the slip-radio doesn't switch to receiving mode, when the problem occurs.

BaN-Co, 2016-12-10:
I've checked the network traffic with the sensniff when the problem occurs and here is what I see:
- The slip-radio sends the broadcast messages successfully.
- The Node1 and Node2 can communicate with each other without any problems.
- The slip-radio doesn't send ACKs to the messages from the Node1 to it.
Can't upload the wireshark log right now due to the strange github error, I'll try to do it latter.

BaN-Co, 2016-12-11:
I've finally traced to the sourse of the problem.
In the file ./cpu/cc26xx-cc13xx/rf-core/prop-mode.c there is a request() function declaration, which is registed to be called by the LPM module to know the MAX supported PM mode of the application. This function is called periodically and it checks the ready status of the RF Core (HWREG(PRCM_BASE + PRCM_O_PDSTAT1RFC) & PRCM_PDSTAT1RFC_ON)), and if the RF Core is not ready, the request() function will return the LPM_MODE_DEEP_SLEEP mode instead of LPM_MODE_SLEEP and this breaks the slip-radio receiving and continue to switch to the LPM_MODE_DEEP_SLEEP afterwards.
This happens only after the specific conditions or maybe it's just a streak of coincidences, but I've seen this 5 times in a row. I can describe these conditions, but these conditions doesn't allways lead to the appearance of this problem, sometimes, when the conditions will meet, the problem will happen, but sometimes with these same conditions everything will work fine.
I don't know why this happens, maybe voltage drop?
I'll try to check some more.

BaN-Co, 2016-12-12:
Here is a temporary fix for this problem:
https://github.com/BaN-Co/contiki/commit/9d222f7c8b52bc1c3ec40130847b035acc0f9818

arurke, 2016-12-13:
@BaN-Co I'll just note that the symptoms (missing ACKs) you are seeing are similar to what I saw in #1878 . Have you tried the latest workaround proposed there (8 slots in rx buffer)? (however it looks you might be closer to the core of the problem)

BaN-Co, 2016-12-14:
@arurke Thank you for pointing it out, I've tested with these fixes (increased buffers up to 8 + there is missing code for new buffers in the off() function in this commit: https://github.com/g-oikonomou/contiki/commit/ee1f5113abf9a761035cf4856ecb2db4453019b3) and there is no problems so far, but I'll keep testing some more.
Also, seems like the the PROP_ERROR_RXBUF error is causing the RF core domain to turn off.

g-oikonomou, 2016-12-19:
Closing as duplicate of #1878

closed 2016-12-19 by g-oikonomou