contiki-os/contiki #1983: Debuger interface Locked for cc2650

MohsenBiglari, 2016-12-08:
Hi
I'm using Contiki for Sensortag cc2650 but when I want to program .bin file into Sensortag with Flash programmer 2 i see this error :

>Initiate access to target: XDS-L3004173.
>Failed to read target memory.
>Error while reading memory range. Address 0x40091090.
>Debug interface is locked. Aborting action. Only forced  mass erase is possible.

Any body can help me?

msloth, 2016-12-08:
In Smartrf flash programmer, select the CPU, press the wrench icon and
choose cc26xx forced erase. Done. There are settings in the CCA to lock
down the JTAG so that no-one can read out the firmware (that way).

msloth, 2016-12-08:
But, to take the larger perspective, this isn't an issue with Contiki per se, thus your question is off topic here and would be better suited at TI e2e forum.

MohsenBiglari, 2016-12-08:
Hi msloth
I did your suggestion but I can't lock down the JTAG interface.
Also I searched all posts in TI e2e forum and no body can solved this problem.
Thanks alot

jonnteolsson, 2016-12-10:
Please post on this forum: https://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/

closed 2016-12-15 by g-oikonomou