contiki-os/contiki #1985: Bug in Contiki CoAP Observing  implementation with CON messages

Altamura, 2016-12-10:
Hello to everybody. 
I have found some bug in the Contiki CoAP code, more specifically in the portion regardind the Observing functionality. I have simulated a network in Cooja (with Z1 motes) including:
- a common border router
- a CoAP server with an observable resource (I take the example "er-example-server.c" in the /er-rest-example folder). The observable resource is the /test/push resource.
- a CoAP client that makes a request of observing to the observable resource of the server (I take the "er-example-observe-client.c" in the same folder cited above). In that code you should modify the server ip address accordingly to the ip address used in the simulation.

In order to correctly compile and execute the code I have added the following parameters to the project-conf.h file:
#undef NBR_TABLE_CONF_MAX_NEIGHBORS
#define NBR_TABLE_CONF_MAX_NEIGHBORS     10
#undef UIP_CONF_MAX_ROUTES
#define UIP_CONF_MAX_ROUTES   10
#undef UIP_CONF_BUFFER_SIZE
#define UIP_CONF_BUFFER_SIZE    280

In the simulation everything works fine and the client correctly received the periodic notifications from the server. Each notification consist of a string "VERY LONG EVENT X" where X is an integer incremented for each step.

However this approach uses NON message, so when the client receives a notification he dosn't reply with an ACK. I have modified this behavior changing the 'coap_notify_observers_sub()' function in /apps/er-coap/er-coap-observe.c with the instruction:

coap_init_message(notification, COAP_TYPE_CON, CONTENT_2_05, 0);

In this way the server notify each subscribers for that resource with a CON message. From this moment the observing mechanism doesn't work as expected infact the first character in the notification is always different and wrong. Infact the client correctly receive the notification but its payload is (wrongly) changed when he reply with the ACK. More in detail the notification is changed during this instruction:

coap_send_message(addr, port, uip_appdata, len);

in the simple_reply() function (see er-coap-observe-client.c), invoked in the coap_handle_notification() function with:

if(notification->type == COAP_TYPE_CON) {
    simple_reply(COAP_TYPE_ACK, addr, port, notification);
  }

Something wrong happens during the coap_send_message function. What exactly? Thank you for your help.

araldit, 2017-12-28:
I just discovered this bug myself - did you ever investigate any further?

Altamura, 2017-12-28:
No, I'm sorry.

SyedRazaBashir, 2018-04-20:
hi.  how to send cop packets from client to server through border router