contiki-os/contiki #1988: sensortag/cc2650 setup RX timestamp

SoonSYJ, 2016-12-12:
As I know cc2650 support RX buffer timestamp which has been talked on Chapter 23.5.3.1 of "CC13xx, CC26xx SimpleLink Wireless MCU Technical Reference Manual.pdf". I try to implement this feature starting from contiki/examples/rime/example-broadcast.c .
I have added 
`packetbuf_set_attr(PACKETBUF_ATTR_PACKET_TYPE, PACKETBUF_ATTR_PACKET_TYPE_TIMESTAMP);`
before 
`broadcast_send(&broadcast);`

and when we receive a packet 
` printf("RX timestamp %d\n", packetbuf_attr(PACKETBUF_ATTR_TIMESTAMP));`

But the node keep printing "RX timestamp 0". 
Could any please tell me if we can not use packetbuf library on cc26xx chip? And how can I set cc2650 to support RX buffer timestamping. 
Thank you.

SoonSYJ, 2016-12-12:
I have checked the file "contiki/cpu/cc26xx-cc13xx/lib/cc26xxware/driverlib/rf_common_cmd.h". And found that this is different from the official driver library provided by TI. The CMD_RX has been modified to CMD_RX_TEST and the timestamp attribute has been removed. 
So my question is why we have to do this? Why we can use timestamp command here?
Please any one give me some advice. Thank you so much.

SoonSYJ, 2016-12-12:
I have checked the file "contiki/cpu/cc26xx-cc13xx/lib/cc26xxware/driverlib/rf_common_cmd.h". And found that this is different from the official driver library provided by TI. The CMD_RX and CMD_TX has been removed. So we can get the timestamp attribute.
So my question is why we have to do this? Why we can use timestamp command here?
Please any one give me some advice. Thank you so much.

g-oikonomou, 2016-12-15:
If you look at the CC26xx radio driver, it already implements time stamping for TSCH in IEEE mode. That should be a good starting point for you.

In terms of the packetbuf discussion, I guess that's the same as #1996 and an issue is not a suitable place to have this conversation anyway.

closed 2016-12-15 by g-oikonomou