contiki-os/contiki #1990: contiki.h is inactive or already defined

juttTechnology, 2016-12-13:
hello
        i am using visualgdb with nrf52.
i import all required files into the visualgdb and i am also using my own makefile. 
i try to run the nrf52 blink example. everything is working and i also can debug the code.
 no Target is passing because i only have specific files regarding to nrf52.
i have one question.

the contiki.h is in active means he behave like this is already defined but where ?
i also check the contiki makefiles but i found nothing.
this is something look like preprocessores. but i have nothing special in my preprocessors regarding to  contiki.h.
can any one guide me to identify this issue.

thank you

greg-king5, 2016-12-14:
Three statements in "Makefile.include" build the path to the location of "contiki.h":
1. https://github.com/contiki-os/contiki/blob/master/Makefile.include#L105-L106
2. https://github.com/contiki-os/contiki/blob/master/Makefile.include#L184-L185
3. https://github.com/contiki-os/contiki/blob/master/Makefile.include#L190

The result is that the compiler is told:  `-I$(CONTIKI)/core/.`

juttTechnology, 2016-12-14:
thank you..
i fix the makefile setting in the visualgdb. Now its Working..

closed 2016-12-14 by juttTechnology