contiki-os/contiki #1991: Cooja native simulation mapping fails

ryankurte, 2016-12-13:
Tested with local cooja(OSX) as well as instant contiki (Linux).

Cooja mote configuration wizard fails on step 3/4: Library Memory Addresses.

[image]

[image]

Compilation of nodes for the Cooja target works, then creation causes a java error.

[image]

[image]

afonsomota, 2017-01-03:
I've been curious about the same topic. 

Considering the instant contiki case from what I have seen, the problem is (at least) in:
  - ConfigurationWizard.java: lines 864 and 869 should be _END instead of _SIZE to be the same as in external_tools.config
  - Some regex in external_tools.config do not yield any result in parsing the output of nm. For instance, in COMMAND_DATA_END, it should be D and not A

This, in terms of memory section sizes. The problem with variables is another.

I have successfully obtained the sections right using also command based address parsing. I can do a Pull Request when I have this working fully.

Cheers,

Afonso

casaroli, 2017-01-03:
I have it working with java 32bit.