contiki-os/contiki #1992: Slip-radio and commands checksum

BaN-Co, 2016-12-14:
Why is there no checksum verification in the communication between slip-radio and 6lbr?
If the communication is over USB, then there shouldn't be any problem, but if it's over UART, then there will be a lot of problems.
The last straw was that the slip-radio sent out the corrupted ICMP packet and when the node received it, instead of processing the single RPL option, it processed a few PAD1 options + removing RPL extension header and the node hanged in result. And there was also one more time when the node received corrupted ICMP packet and when the node received it, instead of processing the single RPL option, it processed the PADN option + removing RPL extension header and the node hanged in result.
Though it's not a certain that it hanged because of this packet, but it was the last thing this node did.
I think we need to add the checksum to the commands sent between the slip-radio and 6lbr.

joakimeriksson, 2016-12-14:
There is a completely new version of the SLIP radio and "NBR" that is fully Contiki-based and includes checksums and lots of new features. We would be happy to upstream this if we get some people to tell us that it would be of high value (it is all open source but includes the complete Yanzi application layer). Take a look at sics-iot/sparrow - there is some documentation on the Wiki there but it is brief so please give feedback if you thing something is missing. The serial radio is under products/sparrow-serial-radio and the border-router is under products/sparrow-border-router.

arurke, 2016-12-19:
Absolutely interested if it is compatible for use with 6lbr

BaN-Co, 2016-12-19:
@joakimeriksson
Thank you for your suggestion, it's quite interesting, especially the OTA functionality. I'll try to port it to the cc26xx-cc13xx cpu later, but first I want to make the current configuration work stable and then I'll try to improve it.
I've added the checksum to the slip-radio and 6lbr code, it'll make do for now.
@arurke 
No, it woks only with the custom sparrow-border-router.