contiki-os/contiki #1996: packetbuf_set_attr() bug

SoonSYJ, 2016-12-14:
Add following codes to examples/rime/example-broadcast.c
```
packetbuf_set_attr(PACKETBUF_ATTR_PACKET_TYPE, PACKETBUF_ATTR_PACKET_TYPE_TIMESTAMP);
printf("Type %hu\n", packetbuf_attr(PACKETBUF_ATTR_PACKET_TYPE));
```
Download the image to a sensortag/cc2650 board. And the serial output is "Type 0". As I understand this must be "Type 4".

Please anyone give some advice. Thank you.

BaN-Co, 2016-12-14:
Check the assembly output first:
make  TARGET=... BOARD=...  example-broadcast.lst

SoonSYJ, 2016-12-15:
@BaN-Co Thank you. I have done the build as you have said. But I am not sure how to use assembly output to debug. Is there any documents about assembly output? Anyway I try to paste some parts of example-broadcast.lst. May be you would like to have a look. 
0000a780 <autostart_processes>:
 ```
   a780:	0428 2000 0000 0000 6568 6c6c 006f 0000     (.. ....hello...
    a790:	7954 6570 2520 7568 000a 0000 7262 616f     Type %hu....broa
    a7a0:	6364 7361 2074 656d 7373 6761 2065 6572     dcast message re
    a7b0:	6563 7669 6465 6620 6f72 206d 6425 252e     ceived from %d.%
    a7c0:	3a64 2720 7325 0a27 0000 0000 5352 4953     d: '%s'.....RSSI
    a7d0:	725f 6d69 3a65 2520 0a64 0000 6954 656d     _rime: %d...Time
    a7e0:	7473 6d61 3a70 2520 0a75 0000 694c 7473     stamp: %u...List
    a7f0:	6e65 7420 6d69 3a65 2520 0a75 0000 0000     en time: %u.....
    a800:	7242 616f 6364 7361 2074 7865 6d61 6c70     Broadcast example
    a810:	0065 0000                                                             e...
```
And there is an error here. But I think that it is the reason we do not connect our node.
```
0000aba8 <nullsec_driver>:
    aba8:	aba0 0000 7459 0000 7469 0000 745d 0000     ....Yt..it..]t..
    abb8:	ff90 00ff 6672 635f 726f 5f65 6573 646e     ....rf_core_send
    abc8:	635f 646d 203a 7830 3025 6c38 2078 6954     _cmd: 0x%08lx Ti
    abd8:	656d 756f 0a74 0000 6672 635f 726f 5f65     meout...rf_core_
    abe8:	6573 646e 635f 646d 203a 4652 7720 7361     send_cmd: RF was
    abf8:	6f20 6666 0000 0000 6672 635f 726f 5f65      off....rf_core_
    ac08:	6f70 6577 5f72 7075 203a 4d43 5f44 4950     power_up: CMD_PI
    ac18:	474e 6620 6961 2c6c 4320 444d 5453 3d41     NG fail, CMDSTA=
    ac28:	7830 3025 6c38 0a78 0000 0000 6672 635f     0x%08lx.....rf_c
    ac38:	726f 5f65 6567 5f74 6172 5f74 7472 5f63     ore_get_rat_rtc_
    ac48:	666f 7366 7465 203a 5953 434e 535f 4154     offset: SYNC_STA
    ac58:	5452 525f 5441 6620 6961 2c6c 4320 444d     RT_RAT fail, CMD
    ac68:	5453 3d41 7830 3025 6c38 0a78 0000 0000     STA=0x%08lx.....
    ac78:	6672 635f 726f 5f65 6d63 5f64 6b6f 203a     rf_core_cmd_ok: 
    ac88:	5953 434e 535f 4154 5452 525f 5441 7720     SYNC_START_RAT w
    ac98:	6961 2c74 4320 444d 5453 3d41 7830 3025     ait, CMDSTA=0x%0
    aca8:	6c38 2c78 7320 6174 7574 3d73 7830 3025     8lx, status=0x%0
    acb8:	7834 000a 6672 635f 726f 5f65 6567 5f74     4x..rf_core_get_
    acc8:	6172 5f74 7472 5f63 666f 7366 7465 203a     rat_rtc_offset: 
    acd8:	5953 434e 535f 4f54 5f50 4152 2054 6166     SYNC_STOP_RAT fa
    ace8:	6c69 202c 6572 2074 6425 4320 444d 5453     il, ret %d CMDST
    acf8:	3d41 7830 3025 6c38 0a78 0000 6672 635f     A=0x%08lx...rf_c
    ad08:	726f 5f65 6d63 5f64 6b6f 203a 5953 434e     ore_cmd_ok: SYNC
    ad18:	535f 4f54 5f50 4152 2054 6177 7469 202c     _STOP_RAT wait, 
    ad28:	4d43 5344 4154 303d 2578 3830 786c 202c     CMDSTA=0x%08lx, 
    ad38:	7473 7461 7375 303d 2578 3430 0a78 0000     status=0x%04x...
    ad48:	7366 705f 776f 7265 6f64 6e77 203a 4d43     fs_powerdown: CM
    ad58:	5344 4154 303d 2578 3830 786c 000a 0000     DSTA=0x%08lx....
    ad68:	7366 705f 776f 7265 6f64 6e77 203a 4d43     fs_powerdown: CM
    ad78:	5344 4154 303d 2578 3830 786c 202c 7473     DSTA=0x%08lx, st
    ad88:	7461 7375 303d 2578 3430 0a78 0000 0000     atus=0x%04x.....
    ad98:	6672 635f 726f 5f65 6f62 746f 203a 6672     rf_core_boot: rf
    ada8:	635f 726f 5f65 6f70 6577 5f72 7075 2928     _core_power_up()
    adb8:	6620 6961 656c 0064 6672 635f 726f 5f65      failed.rf_core_
    adc8:	6f62 746f 203a 6672 635f 726f 5f65 7473     boot: rf_core_st
    add8:	7261 5f74 6172 2874 2029 6166 6c69 6465     art_rat() failed
    ade8:	0000 0000 6672 635f 726f 5f65 6572 7473     ....rf_core_rest
    adf8:	7261 5f74 6172 3a74 7220 5f66 6f63 6572     art_rat: rf_core
    ae08:	735f 6f74 5f70 6172 2874 2029 6166 6c69     _stop_rat() fail
    ae18:	6465 0000 6672 635f 726f 5f65 6572 7473     ed..rf_core_rest
    ae28:	7261 5f74 6172 3a74 7220 5f66 6f63 6572     art_rat: rf_core
    ae38:	735f 6174 7472 725f 7461 2928 6620 6961     _start_rat() fai
    ae48:	656c 0064 6573 7574 5f70 6e69 6574 7272     led.setup_interr
    ae58:	7075 7374 203a 6f4e 6120 6363 7365 0073     upts: No access.
    ae68:	4652 4520 7272 726f 0000 0000 6672 635f     RF Error....rf_c
    ae78:	726f 5f65 6f70 6577 5f72 7075 2928 6620     ore_power_up() f
    ae88:	6961 656c 0064 0000 4652 203a 6142 2064     ailed...RF: Bad 
    ae98:	5243 0043 4652 4920 5253 6320 6c61 656c     CRC.RF ISR calle
    aea8:	2064 7562 2074 4652 6e20 746f 7220 6165     d but RF not rea
    aeb8:	7964 2e2e 202e 4150 494e 2143 0021 0000     dy... PANIC!!...
    aec8:	4343 3331 7878 2f20 4320 3243 7836 2078     CC13xx / CC26xx 
    aed8:	4652 6420 6972 6576 0072 0000               RF driver...
```

BaN-Co, 2016-12-15:
Where in the code did you placed the printf code?
I've tried it on my board (custom CC1310-based), I've placed that code in the PROCESS_THREAD(example_broadcast_process, ev, data), before the while(1) loop and I've got the "Type 4" output.
Here is what I have in the assembler output:
```
     172:	2104      	movs	r1, #4
     174:	200d      	movs	r0, #13
     176:	f004 fcb9 	bl	4aec <packetbuf_set_attr>
     17a:	200d      	movs	r0, #13
     17c:	f004 fcbe 	bl	4afc <packetbuf_attr>
     180:	4601      	mov	r1, r0
     182:	4812      	ldr	r0, [pc, #72]	; (1cc <process_thread_example_broadcast_process+0x88>)
     184:	f002 fa5c 	bl	2640 <printf>
```

SoonSYJ, 2016-12-15:
Thank you. That's cool. Here are my assemble output. And I think there are something wrong. How do you think about this.
```
    1030:	2104      	movs	r1, #4
    1032:	200d      	movs	r0, #13
    1034:	f005 fba0 	bl	6778 <packetbuf_set_attr>
    1038:	2106      	movs	r1, #6
    103a:	480a      	ldr	r0, [pc, #40]	; (1064 <process_thread_example_broadcast_process+0x84>)
    103c:	f005 fb40 	bl	66c0 <packetbuf_copyfrom>
    1040:	200d      	movs	r0, #13
    1042:	f005 fba1 	bl	6788 <packetbuf_attr>
    1046:	4601      	mov	r1, r0
    1048:	4807      	ldr	r0, [pc, #28]	; (1068 <process_thread_example_broadcast_process+0x88>)
    104a:	f003 f989 	bl	4360 <printf>
    104e:	4804      	ldr	r0, [pc, #16]	; (1060 <process_thread_example_broadcast_process+0x80>)
    1050:	f004 fae8 	bl	5624 <broadcast_send>
    1054:	e7db      	b.n	100e <process_thread_example_broadcast_process+0x2e>
```

SoonSYJ, 2016-12-15:
@BaN-Co 
Thank you. That's cool. Here are my assemble output. And I think there are something wrong. How do you think about this.
```
    1030:	2104      	movs	r1, #4
    1032:	200d      	movs	r0, #13
    1034:	f005 fba0 	bl	6778 <packetbuf_set_attr>
    1038:	2106      	movs	r1, #6
    103a:	480a      	ldr	r0, [pc, #40]	; (1064 <process_thread_example_broadcast_process+0x84>)
    103c:	f005 fb40 	bl	66c0 <packetbuf_copyfrom>
    1040:	200d      	movs	r0, #13
    1042:	f005 fba1 	bl	6788 <packetbuf_attr>
    1046:	4601      	mov	r1, r0
    1048:	4807      	ldr	r0, [pc, #28]	; (1068 <process_thread_example_broadcast_process+0x88>)
    104a:	f003 f989 	bl	4360 <printf>
    104e:	4804      	ldr	r0, [pc, #16]	; (1060 <process_thread_example_broadcast_process+0x80>)
    1050:	f004 fae8 	bl	5624 <broadcast_send>
    1054:	e7db      	b.n	100e <process_thread_example_broadcast_process+0x2e>
```

SoonSYJ, 2016-12-15:
@BaN-Co Here is the code
PROCESS_THREAD(example_broadcast_process, ev, data)
{
  static struct etimer et;

  PROCESS_EXITHANDLER(broadcast_close(&broadcast);)
  PROCESS_BEGIN();

  broadcast_open(&broadcast, 129, &broadcast_call);
  while(1) {

    /* Delay 2-4 seconds */
    etimer_set(&et, CLOCK_SECOND * 4 + random_rand() % (CLOCK_SECOND * 4));
    PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&et));

    packetbuf_set_attr(PACKETBUF_ATTR_PACKET_TYPE,
                       PACKETBUF_ATTR_PACKET_TYPE_TIMESTAMP);

    packetbuf_copyfrom("hello", 6);
    printf("Type %hu\n", packetbuf_attr(PACKETBUF_ATTR_PACKET_TYPE));

    broadcast_send(&broadcast);
  }

  PROCESS_END();
}

SoonSYJ, 2016-12-15:
@BaN-Co 
I have found the problem. I should read PACKETBUF_ATTR_PACKET_TYPE before pakcetbuf_copyfrom(). Because the function has run packetbuf_clear( ). I think I have miss understanding packetbuf_attr as some kind of registers. Actually these are variables which is used sharing datas between processes. Am I right? By the way, are there any documents talking about the functions or APIs. I found that I can only read the code and try to understand them. Some documents will help a lot. 
Thank you.

BaN-Co, 2016-12-15:
@SoonSYJ Check the source code first, the packetbuf_copyfrom() clears all the attributes with the packetbuf_attr_clear() function in the packetbuf_clear() function.

BaN-Co, 2016-12-15:
About understanding the basics of the packetbuf and rime, you can search it in google, for example, here:
http://anrg.usc.edu/contiki/index.php/Packetbuffer_Basics
http://dunkels.com/adam/dunkels07adaptive.pdf

SoonSYJ, 2016-12-15:
@BaN-Co Thank you. I have solved this problem. 
I am working on time synchronisation with sensortag/cc2650. Actually Contiki has a lib called timesynch@{contiki/core/net/rime/}. But I did not find any documents about the workflow. As I have just met Contiki for one week, and it really take lots of time to understand it. I think there should be document talks about these basic lib. 
I have read the work of USC and the paper about rime. But I have got little help. Maybe my English is too poor to understand it well. I will write down my understanding and try to post them some well. 
Thank you very much. Thank you very much. Thank you very much.

closed 2016-12-15 by g-oikonomou