contiki-os/contiki #2005: TSCH get timestamp from the packet with cc2650 chip

SoonSYJ, 2016-12-19:
Time synchronisation is embedded in TSCH MAC library. The basic idea behind time synchronisation is exchanging radio timestamps between nodes. After reading the code about TSCH, I have got a confusion.
In file "tsch-slot-operation.c", a node get the RX time when a packet is received by calling;
`        NETSTACK_RADIO.get_object(RADIO_PARAM_LAST_PACKET_TIMESTAMP, &rx_start_time, sizeof(rtimer_clock_t));`
As I am using cc2650 chip in IEEE mode, the **_NETSTACK_RADIO_** protocol is linked to ieee_mode_driver @{contiki/cpu/cc26xx-23xx/rf-core/ieee-mode.c}. By calling the **_get_object_**() function, I can get the **_last_packet_timestamp_** variable. And _**last_packet_timestamp**_ is updated by the read_frame() function when a proper packet is received.
Here I find that the radio relevant params such as rssi, lqi, payload and timestamp are extracted from a buffer called **_rx_read_entry_**. 
## And I can not understand:
1. How the radio params is passed to this buffer?
2. What's the structure of this buffer? I mean that the position and length of each element in this buffer. I have found a figure from the cc13xx cc26xx reference manual, but it seems not right. 
[image]
Please anyone give me some advice

g-oikonomou, 2016-12-19:
https://github.com/contiki-os/contiki/wiki#help

closed 2016-12-19 by g-oikonomou

SoonSYJ, 2016-12-20:
No one would like to give help. No one like to share experience. So mean.