contiki-os/contiki #2006: CC2650 reboot after a while

naguirre, 2016-12-19:
Hi,
I tried to get cc2650 sensortag working with master #9ef28c7
I'm using rpl-border-router without any modifications, and i'm using tunslip6 like this : 
sudo tools/tunslip6 -B 115200 -I -L -s /dev/ttyACM0 -v5 fd00::1/64
When i start the node running cc26xx-web-demo the green led flash quickly and then slower. I can see the node ip on the rpl-border-router web page. And i'm able to ping6 this IP. but after a while (~1minute) the node resets (red and green leds) 

Are you aware of this problem ?

naguirre, 2016-12-19:
I tried by disabling mqtt and the node doesn't reset anymore.

g-oikonomou, 2016-12-19:
Could you be looking at #1858?

naguirre, 2016-12-19:
I can confirm i get the same behaviour. I i keep the code as is, the node reboot : 
9ef28c7
CC2650Sensortag 
REV 1.3
And debugguer DevPack attached

If i change the code with this patch, the code keep working (uptime : 2713 seconds) 
```diff
diff --git a/examples/cc26xx/cc26xx-web-demo/project-conf.h b/examples/cc26xx/cc26xx-web-demo/project-conf.h
index adf2f14..61db374 100644
--- a/examples/cc26xx/cc26xx-web-demo/project-conf.h
+++ b/examples/cc26xx/cc26xx-web-demo/project-conf.h
@@ -37,10 +37,10 @@
 #define RF_BLE_CONF_ENABLED                   1
 /*---------------------------------------------------------------------------*/
 /* Enable/Disable Components of this Demo */
-#define CC26XX_WEB_DEMO_CONF_MQTT_CLIENT      1
+#define CC26XX_WEB_DEMO_CONF_MQTT_CLIENT      0
 #define CC26XX_WEB_DEMO_CONF_6LBR_CLIENT      1
 #define CC26XX_WEB_DEMO_CONF_COAP_SERVER      1
-#define CC26XX_WEB_DEMO_CONF_NET_UART         1
+#define CC26XX_WEB_DEMO_CONF_NET_UART         0
 /*---------------------------------------------------------------------------*/
 /* Enable the ROM bootloader */
 #define ROM_BOOTLOADER_ENABLE                 1

```

g-oikonomou, 2017-01-10:
#1858

closed 2017-01-10 by g-oikonomou