contiki-os/contiki #2007: How reduce the RAM data size

juttTechnology, 2016-12-20:
i am using the simple_udp_rpl example.
i just simple add the all the required files into the visualgdb.
         Used FLASH: 172KB out of 512KB (33%)
         Used RAM: 36KB out of 64KB (57%) [+2280]
i am trying to reduce the RAM data from 36kb to 25kb.

here is my Symbol Table

SYMBOL TABLE:
20002a54 l    d  .bss   00000000 .bss
20002a54 l       .bss   00000001 completed.7620
20002a58 l       .bss   00000018 object.7625
20006dd4 l     O .bss   00000004 heap_end.5213
20002ca0 l     O .bss   00000010 registrations_memb_count
20002aa0 l     O .bss   00000200 registrations_memb_mem
20002a70 l     O .bss   00000004 own_services_list
20002a74 l     O .bss   00000004 others_services_list
20002a78 l     O .bss   00000001 started
20002a7c l     O .bss   00000010 sendtimer
20002a8c l     O .bss   00000004 outconn.11032
20002a90 l     O .bss   00000010 periodic.11031
20002cb0 l     O .bss   00000001 leds
20002cb4 l     O .bss   00000004 ptr.8072
20002cb8 l     O .bss   00000008 rxbuf
20002cc0 l     O .bss   00000080 buf.8071
20002d40 l     O .bss   00000001 overflow.8065
20002d41 l     O .bss   00000080 rxbuf_data
20002dc4 l     O .bss   00000004 i.8087
20002dc8 l     O .bss   00000001 num_sensors
20002dcc l     O .bss   00000004 events.8088
20002dd0 l     O .bss   00000004 callback
20002dd4 l     O .bss   00000001 uncomp_hdr_len
20002dd5 l     O .bss   00000001 packetbuf_hdr_len
20002dd8 l     O .bss   00000004 packetbuf_ptr
20002ddc l     O .bss   00000004 last_rssi
20002de0 l     O .bss   00000190 _ds6_neighbors_mem
20003154 l     O .bss   0000000a routememb_memb_count
20002fc4 l     O .bss   00000190 routememb_memb_mem
20002fc0 l     O .bss   00000002 defaultroutermemb_memb_count
20002f80 l     O .bss   00000040 defaultroutermemb_memb_mem
200031b0 l     O .bss   0000000a neighborroutememb_memb_count
20003160 l     O .bss   00000050 neighborroutememb_memb_mem
200031bc l     O .bss   00000050 _nbr_routes_mem
20002f70 l     O .bss   00000004 notificationlist_list
20002f74 l     O .bss   00000004 defaultrouterlist_list
20002f78 l     O .bss   00000004 num_routes
20002f7c l     O .bss   00000004 routelist_list
2000320c l     O .bss   00000004 locprefix
20003210 l     O .bss   00000004 locaddr
20003214 l     O .bss   00000004 locmaddr
20003218 l     O .bss   00000010 loc_fipaddr
20003228 l     O .bss   00000001 rscount
2000322c l     O .bss   00000004 echo_reply_callback_list_list
20003230 l     O .bss   00000010 tmp_ipaddr
20003240 l     O .bss   00000004 input_handler_list_list
20003244 l     O .bss   00000004 nd6_opt_llao
20003248 l     O .bss   00000001 nd6_opt_offset
2000324c l     O .bss   00000004 nbr
20003250 l     O .bss   00000004 nd6_opt_prefix_info
20003254 l     O .bss   00000004 prefix
20003258 l     O .bss   00000010 ipaddr
20003268 l     O .bss   00000004 addr
2000326e l     O .bss   00000002 lastport
20003270 l     O .bss   00000004 iss
20003275 l     O .bss   00000500 databuffer
20003775 l     O .bss   00000001 started
20003778 l     O .bss   00000010 periodic
20003788 l     O .bss   00000004 outputfunc
2000378c l     O .bss   000000a4 s
20003830 l     O .bss   00000078 _link_stats_mem
[... 110 lines cut ...]
20008850 g     O .bss   00000004 uip_appdata
20008854 g     O .bss   00000004 uip_conn
20008bac g     O .bss   00000001 tcpip_event
20008858 g     O .bss   000001e0 uip_conns
20008a38 g     O .bss   00000008 uip_icmp6_conns
20003274 g     O .bss   00000001 uip_ext_bitmap
20008a40 g     O .bss   00000028 uip_listenports
200092ec g     O .bss   00000004 m_line_num
20008da0 g     O .bss   00000030 energest_current_time
200069e0 g     O .bss   00000004 process_current
20008320 g     O .bss   00000010 uip_ds6_timer_rs
2000326c g     O .bss   00000001 uip_ext_len
200092de g     O .bss   00000006 sensors_flags
200081e8 g     O .bss   00000001 serial_line_event_message
2000326d g     O .bss   00000001 uip_ext_opt_offset
20008dd1 g     O .bss   00000001 ble_event_interface_added
20006dd0 g     O .bss   00000004 __malloc_sbrk_start
200081e9 g     O .bss   00000001 sensors_event
20006dcc g     O .bss   00000004 __malloc_free_list
20008bad g     O .bss   00000001 tcpip_icmp6_event
20008a68 g     O .bss   00000002 uip_slen
20008a6c g     O .bss   00000140 uip_udp_conns
20008dd2 g     O .bss   0000050c input_packet

any idea, how this can be achiveable ?

darkopetrovic, 2016-12-23:
Set a lower value for the following parameter in your `project-conf.h` file: 

    #define UIP_CONF_BUFFER_SIZE 1300

g-oikonomou, 2017-01-10:
There are plenty of discussions about this on the mailing list and there is also this: https://github.com/contiki-os/contiki/wiki/Reducing-Contiki-OS-firmware-size

closed 2017-01-10 by g-oikonomou