contiki-os/contiki #2010: CC1310 don’t connect to 6lbr

Alexor86, 2016-12-21:
Hello!
I have CC1310 LaunchPad with SLIP firmware connected to BeagleBone and CC1310  LaunchPad with cc26xx-web-demo. In SLIP project-conf.h I commented out QUEUEBUF_CONF_NUM and UIP_CONF_BUFFER_SIZE and added RF_CORE_CONF_CHANNEL 25
UART start outputs:
SLIP:
> Starting Contiki-3.x-3064-g9ef28c7 With DriverLib v0.47020 TI CC1310 LaunchPad
> IEEE 802.15.4: No, Sub-GHz: Yes, BLE: No, Prop: No Net: slipnet
>  MAC: CSMA  RDC: ContikiMAC, Channel Check Interval: 16 ticks
> Channel 25 Node ID 9690
> Slip Radio started...
cc26xx-web-demo:
> Starting Contiki-3.x-3064-g9ef28c7 With DriverLib v0.47020 TI CC1310 LaunchPad
> IEEE 802.15.4: No, Sub-GHz: Yes, BLE: No, Prop: No Net: sicslowpan
>  MAC: CSMA RDC: ContikiMAC, Channel Check Interval: 16 ticks
> Channel 25 Node ID: 60151
> CC26XX Web Demo Process CC26XX Web Server CC26XX CoAP Server
> 6LBR Client Process CC26XX MQTT Client Process CC26XX Net UART Process

6lbr starts well:

> ********************************************************************************
> Wed Dec 21 15:50:36 UTC 2016 : Starting 6LBR
> /usr/lib/6lbr/bin/cetic_6lbr_router -c /etc/6lbr/nvm.dat -o /etc/6lbr/nvm.conf -s /dev/ttyS4 -r -t eth0 -B 115200 -U /usr/lib/6lbr/6lbr-ifup -D /usr/lib/6lbr/6lbr-ifdown -w /usr/lib/6lbr/www -L 3  -W /var/log/6lbr.timestamp -P 60 -C /var/log/6lbr.ip  -m /usr/lib/6lbr/plugins   
> Contiki 3.0 started with IPV6, RPL
> Rime started with address 1.2.3.4.5.6.7.8
> MAC mac_wrapper RDC br-rdc SEC llsec-wrapper NETWORK sicslowpan
> 2016-12-21 15:50:36.060677: NOTICE: 6LBR: Starting 6LBR version 1.5.x (Contiki 3.0)
> 2016-12-21 15:50:36.061317: INFO: ETH: 6LBR watchdog started (interval: 60)
> 2016-12-21 15:50:36.061562: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/lwm2m.so
> 2016-12-21 15:50:36.061989: INFO: PLUGIN: Initialising lwm2m-client
> 2016-12-21 15:50:36.062019: INFO: PLUGIN: Loading /usr/lib/6lbr/plugins/dummy.so
> 2016-12-21 15:50:36.062294: INFO: NVM: NVM Magic : 2009
> 2016-12-21 15:50:36.062325: INFO: NVM: NVM Version : 5
> 2016-12-21 15:50:36.062355: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
> 2016-12-21 15:50:36.062386: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
> 2016-12-21 15:50:36.062447: NOTICE: 6LBR: Log level: 30 (services: ffffffff)
> 2016-12-21 15:50:36.062447: INFO: MAC: Using 'csma' mac driver
> 2016-12-21 15:50:36.070961: INFO: SLIP: SLIP started on /dev/ttyS4
> 2016-12-21 15:50:36.081520: INFO: SCMD: Started br-cmd process
> 2016-12-21 15:50:36.081581: INFO: BR-RDC: Reset SLIP Radio
> 2016-12-21 15:50:36.081612: INFO: BR-RDC: Fetching MAC address
> 2016-12-21 15:50:36.085762: ERROR: SLIP: Unknown command
> 2016-12-21 15:50:36.085915: INFO: BR-RDC: Got MAC: 00:12:4b:00:0e:09:25:da
> 2016-12-21 15:50:37.081490: INFO: ETH: RAW Ethernet interface init
> 2016-12-21 15:50:37.095467: INFO: TAP: opened device /dev/eth0
> 2016-12-21 15:50:37.095619: INFO: TAP: Running 6lbr-ifup script '/usr/lib/6lbr/6lbr-ifup'
> eth0      Link encap:Ethernet  HWaddr 20:cd:39:dd:9a:27  
>           inet addr:169.254.2.4  Bcast:169.254.255.255  Mask:255.255.0.0
>           UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
>           RX packets:5419 errors:0 dropped:25 overruns:0 frame:0
>           TX packets:3863 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:2967479 (2.8 MiB)  TX bytes:671777 (656.0 KiB)
> 
> 2016-12-21 15:50:37.164223: INFO: TAP: Eth MAC address : 20:cd:39:dd:9a:27
> 2016-12-21 15:50:37.164314: INFO: ETH: Eth MAC address : 20:cd:39:dd:9a:27
> 2016-12-21 15:50:37.164375: INFO: LLSEC: Using 802.15.4 framer
> 2016-12-21 15:50:37.164406: INFO: LLSEC: Using 'nullsec' llsec driver
> 2016-12-21 15:50:37.164742: ERROR: SLIP: Unknown command
> 2016-12-21 15:50:37.164803: INFO: HTTP: Starting webserver on port 80
> 2016-12-21 15:50:37.165169: INFO: NODECFG: Node Config init
> 2016-12-21 15:50:37.165230: INFO: NODECFG: No node_config.conf file specified
> 2016-12-21 15:50:37.165260: INFO: 6LBR: Tentative local IPv6 address fe80::212:4b00:e09:25da
> 2016-12-21 15:50:37.165321: INFO: 6LBR: Tentative global IPv6 address (WSN) aaaa::212:4b00:e09:25da
> 2016-12-21 15:50:37.165413: INFO: 6LBR: Tentative global IPv6 address (ETH) bbbb::100
> 2016-12-21 15:50:37.165474: INFO: 6LBR: RA Daemon enabled
> 2016-12-21 15:50:37.165474: INFO: 6LBR: Checking addresses duplication
> 2016-12-21 15:50:47.634895: ERROR: PF: Trying to send empty packet
> 2016-12-21 15:50:48.167030: INFO: 6LBR: Configured as DODAG Root aaaa::212:4b00:e09:25da
> 2016-12-21 15:50:48.167122: INFO: 6LBR: Starting as RPL ROUTER
> 2016-12-21 15:50:48.167519: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
> 2016-12-21 15:50:48.167580: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
> 2016-12-21 15:50:48.167641: INFO: UDPS: UDP server started
> CoAP push started
> CoAP server started
> 2016-12-21 15:50:48.167763: INFO: DNS: DNS proxy started
> 2016-12-21 15:50:48.167793: INFO: PLUGIN: Initialising lwm2m-client
> 2016-12-21 15:50:48.167824: INFO: LWM2M: LWM2M Client init
> 2016-12-21 15:50:48.167854: INFO: PLUGIN: Initialising dummy
> 2016-12-21 15:50:48.167885: INFO: DUMMY: Dummy init
> 2016-12-21 15:50:48.167915: INFO: CONFIG: Loading configuration : /etc/6lbr/nvm.conf
> 2016-12-21 15:50:48.167946: WARN: CONFIG: Can not open /etc/6lbr/nvm.conf : No such file or directory
> 2016-12-21 15:50:48.167976: INFO: 6LBR: CETIC 6LBR Started
> RD client started

nvm.dat:

> Channel : 25
> PAN ID : abcd
> 
> WSN network prefix : aaaa::
> WSN network prefix length : 64
> WSN IP address : aaaa::100
> WSN accept RA : True
> WSN IP address autoconf : True
> WSN 6LoWPAN context 0 : aaaa::
> DNS server : ::
> 
> Eth network prefix : bbbb::
> Eth network prefix length : 64
> Eth IP address : bbbb::100
> Eth default router : ::
> Eth IP address autoconf : False
> 
> RA daemon : True
> RA router lifetime : 0
> RA maximum interval : 600
> RA minimum interval : 200
> RA minimum delay : 3
> RA PIO enabled : True
> RA prefix valid lifetime : 86400
> RA prefix preferred lifetime : 14400
> RA RIO enabled : True
> RA RIO lifetime : 1800
> 
> RPL instance ID : 30
> RPL DODAG ID manual : False
> RPL DODAG ID : ::
> RPL DODAG ID global : True
> RPL Preference : 0
> RPL version ID :  : 241
> RPL DIO interval doubling : 8
> RPL DIO minimum interval : 12
> RPL DIO redundancy : 10
> RPL default lifetime : 30
> RPL maximum rank increase : 1792
> RPL minimum rank increment : 256
> RPL lifetime unit : 256
> 
> MAC layer : 1
> Security layer : 0
> Security level : 6
> Security key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
> Noncoresec anti-replay disabled : False
> Noncoresec anti-replay workaround : False
> Filter unknown nodes : False
> 
> NAT 64 Enabled : False
> NAT 64 DHCP Enabled : False
> NAT 64 static port mapping Enabled : False
> NAT 64 RFC-6052 prefix : False
> NAT 64 address : 172.16.0.2
> NAT 64 netmask : 255.255.255.0
> NAT 64 gateway : 172.16.0.1
> 
> Local address rewrite : True
> Smart Multi BR : False
> Webserver configuration page disabled : False
> Webserver port : 80

However on http://[bbbb::100] I don’t see any nodes. Moreover I don’t see CSMA Received packets:
 
[image]

Thanks!

arurke, 2016-12-21:
Not exactly sure what the CSMA stat. is measuring, but for higher layers at least: Contiki changed the default 6lowpan context from aaaa:: to fd00:: some time ago. If you are using a newer version on the nodes I would suggest trying to configure fd00:: on the 6lbr.

Alexor86, 2016-12-22:
@arurke, thanks for your response! I have the latest versions of Contiki.
I changed WSN network settings from aaaa:: to fd00::, but unfortunately it did not help me…
nvm.dat:
> WSN network prefix : fd00::
> WSN network prefix length : 64
> WSN IP address : fd00::100
> WSN accept RA : True
> WSN IP address autoconf : True
> WSN 6LoWPAN context 0 : fd00::
> DNS server : ::

bverdu, 2016-12-22:
CC1310 is a sub-GHz board, you should use a channel from 0 to 10, 25 is for 2.4Ghz frequency.

arurke, 2016-12-22:
I am pretty sure channel 25 is valid, ref. https://github.com/contiki-os/contiki/blob/master/cpu/cc26xx-cc13xx/rf-core/dot-15-4g.h#L92

But a good point - default channel is 0 for prop-mode. So I would verify you have the same channel on node and BR.

bverdu, 2016-12-22:
May be channel 25 is technically valid, but the IEEE 802.14.5 specs allow 1 channel for 868 band (Europe), 10 (extended to 30 in 2006) for 915 band(USA) and 16 channels for 2.4Ghz band, ref: [http://www.radio-electronics.com/info/wireless/ieee-802-15-4/bands-frequencies-radio-air-interface.php](http://www.radio-electronics.com/info/wireless/ieee-802-15-4/bands-frequencies-radio-air-interface.php). I have no issue connecting my two CC1310 launchpads (one with web demo, the other with border-router) using channels 0 to 10, My CC2650 sensortags whith the same platform code in contiki are able to use channels from 11 to 26 with the same setup.

g-oikonomou, 2016-12-22:
In 15.4g: For the band with identifier 4 "863–870 (Europe)", at MR-FSK operating mode 1, the total number of channels is 34. Table 68d, Page 35 in Std 802.15.4g-2012

Alexor86, 2016-12-23:
It seems the problem was in my hardware: I've flashed other launchpad board by the same node firmware and it works well (on channel 25).
However I've got the issue #2015

closed 2016-12-23 by Alexor86