contiki-os/contiki #2015: cc1310 web-demo hangs

vad32m, 2016-12-22:
About week ago I succeed with building my first 6lowPAN sensor network. The setup is following: 6lbr, running under Ubuntu VM with cc1310 slip-radio connected to it. Output from slip-radio is following:
[image].
And another node, running cc26xx-web-demo. Now it have following configuration:
[image]
It worked great, until I downloaded the latest version of Contiki and re-flashed node with cc26xx-web-demo. Unfortunately, I made no backup and I dont know, what commit that was. Flashed with the latest version my node doesnt seem to work: after connecting to the router I try to access web-page on my node, sometimes browser is able to load page, sometimes not, but anyway, after short data transmittion it hangs. Wireshark log from 6lbr interface (my node IP is aaaa::212:4b00:a27:b808):
[image]
After that if I reboot router and slip radio, node wont appear in the 6lbr "sensors" page. Only if I will reboot that node, it will be able to connect to the router. Devices are configured to the same channel and PAN ID, BLE beacon disabled. Will apreciate any help or solution.

BaN-Co, 2016-12-22:
I'd suggest you to enable the debug output for the cc26xx-web-demo node to trace the problem.

jonnteolsson, 2016-12-23:
Suggest to apply this PR and give it a try: https://github.com/contiki-os/contiki/pull/1932
also make sure you are using the correct prefix, contiki core has changed from using aaaa:: so you most probably need to update your 6lbr configuration

vad32m, 2016-12-23:
@jonnteolsson on 6lbr 6LoPWAN context 0 is set to fd00:: . Recomended branch doesnt helped: behaviour is still the same.

vad32m, 2016-12-23:
@BaN-Co are there any easy ways to turn debug on? As far as I see, there are DEBUG compile-time constants in different files, should I change it in each file?

BaN-Co, 2016-12-23:
@vad32m I've just replaced recursively in all files the line:
`#define DEBUG 0
`
to these lines:
```
#ifdef GLOBAL_DEBUG
#define DEBUG 1
#else
#define DEBUG 0
#endif
```
and the line: 
`#define DEBUG DEBUG_NONE
`
to these lines:
```
#ifdef GLOBAL_DEBUG
#define DEBUG DEBUG FULL
#else
#define DEBUG DEBUG_NONE
#endif
```
Then I add this line in the project-conf.h file, if I want to enable the debug output:
`#define GLOBAL_DEBUG
`

g-oikonomou, 2016-12-23:
Take great care with debug statements in the radio driver and contikimac, as they may throw contikimac timings off by a lot

BaN-Co, 2016-12-23:
@g-oikonomou Yes, so I've additionally changed the UART driver to send data with ti_lib_uart_char_put_non_blocking() via interrupts and circular buffer instead of blocking ti_lib_uart_char_put().

vad32m, 2016-12-24:
During experiments noticed strange behavior only with 6lbr, if I launch the same web-demo with rpl-border-router and tunslip6, my demo works OK. Anyway, I modified UART driver, as @BaN-Co suggested, and enclosing my logs. Maybe someone can give me solution. Also I am able to connect my cc26xx-web-demo to the GDB and make some debugging, but I dont know what should I look for.
[WEBREMO_6lbr.txt](https://github.com/contiki-os/contiki/files/671897/WEBREMO_6lbr.txt)

[WEBDEMO_RPL_ROUTER.txt](https://github.com/contiki-os/contiki/files/671898/WEBDEMO_RPL_ROUTER.txt)
[6lbr_logs.zip](https://github.com/contiki-os/contiki/files/671899/6lbr_logs.zip)

vad32m, 2016-12-26:
I examined the log and discovered following:
RPL router substitutes destination address correctly, because from my node log I see following lines:
`15.4-IN:  1 0012:4b00:0af6:0265  0012:4b00:0a27:b808 21 64 (85)
packetbuf_set_len: len 63`
Where `0012:4b00:0a27:b808` is my node's MAC address. 
In case with 6lbr and slip-radio I have following messages:
`15.4-IN:  1 0012:4b00:0af6:0265  0000:0000:0000:0000 15 80 (95)
packetbuf_set_len: len 95`
which is later followed by
`15.4-IN:  1 0012:4b00:0af6:0265  020a:0bff:ff0c:0d0e 21 84 (105)
nullrdc: not for us`
Destination addres is wrong, it is not my nodes address, but MAC of my tap0 interface
[image]
Can anyone give me any advice, maybe there is something wrong with configuration?

greg-king5, 2016-12-27:
> It worked great until I downloaded the latest version of Contiki, and re-flashed the node with cc26xx-web-demo.  Unfortunately, I made no backup; and, I don't know what commit that was.

I assume that you compiled your remote node's previous firmware at the same time that you compiled your slip-radio's code.  If you did, then the previous commit probably was 6cdba10b1d32dfae4e1ba58496855c333b6e3c60 (look at the end of the first line in the slip-radio's greeting).

vad32m, 2016-12-29:
Finally got my WSN working again, but its stability is pretty low. Now my setup is: 6lbr Version : 1.5.x (Contiki-contiki-merge-20160622-77-gdc08685), cc1310 slip-radio flashed with YiKai's firmware(Contiki-3.x-2928-g1ebceb8) from here [this thread](https://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/t/563838). As sensor node I have cc1310 launchpad flashed with web-demo(Contiki-3.x-3067-g406cd56). In project-conf file I disabled BLE beacon and added following line `#define UIP_CONF_ROUTER 0`. Now my router is able to correctly discover that node and I am able to access CoAP and Web-servers. 
[image]. 
6lbr log in this case is following:
[6lbr.log.txt](https://github.com/cetic/6lbr/files/677681/6lbr.log.txt)
But if I will delete line  `#define UIP_CONF_ROUTER 0` (default value is 1 as I think), strange behavior will repeat. And 6lbr will not be able to resolve MAC address. 
[image]
 6lbr log in this case is following:
[6lbr.log.failure.txt](https://github.com/cetic/6lbr/files/677683/6lbr.log.failure.txt)
Are you guys interested in further investigation an may I help you somehow?
BTW lines 
`#ifndef RF_CORE_CONF_CHANNEL`
`#define RF_CORE_CONF_CHANNEL                      25`
`#endif`
are duplicated in file contiki-conf.h for platform srf06-cc26xx.

g-oikonomou, 2017-01-10:
Looks like what you are describing could be a wider issue (with NDP?), not strictly related to this particular platform. This is something we need to keep an eye out for. Meanwhile, just please double-check that you are not merely suffering from #1878, just in case...

>BTW lines
> #ifndef RF_CORE_CONF_CHANNEL
>#define RF_CORE_CONF_CHANNEL 25
> #endif
> are duplicated in file contiki-conf.h for platform srf06-cc26xx.

Pretty sure they are not :)

g-oikonomou, 2017-01-15:
@vad32m this sounds like it may also be related with #1858?

jbartlett12026, 2017-02-01:
@vad32m Hello, I am trying to flash my CC1310 launchpad with the slip-radio and keep getting an incorrect serial output. Did you flash using "make TARGET=srf06-cc26xx BOARD=launchpad/cc1310"? I am trying to get the result you first posted. Thank you

[image]

atiselsts, 2017-02-02:
To the people having problems - it might be that the nodes are having memory corruption issues due to stack overflows (#2089).

vad32m, 2017-02-02:
@jbartlett12026 My workflow is following:
1. Install and configure 6lbr on Linux VM. 
2. Clone git repo to my Windows machine(my build environment) and checkout specific commit, if required.
3. OPTIONALLY: add folder to /platform/srf06-cc26xx/. I have only two launchpads, which is not enough for my experiments, so two more cc1310 custom DIY (HW is the same as launchpad, but without buttons and with single LED) nodes are used in my experiments. If you have natively supported hardware, there is no need in this step. 
4. Modify /platform/srf06-cc26xx/contiki-conf.h and slip-radio's project-conf.h files (both files attached).
5. Build slip-radio with cmd `make TARGET=srf06-cc26xx BOARD=lowpanmod/cc1310 -j8` because now I use custom node for slip-radio (you can use `BOARD=launchpad/cc1310` for your HW).
6. Flash that firmware into cc1310 and get following output:
[image]
7. Connect my slip-radio to Linux VM and launch 6lbr (I removed it from autostart).
8. Compile web-demo for launchpad with cmd `make TARGET=srf06-cc26xx BOARD=launchpad/cc1310 -j8`, do not forget to change `#define RF_BLE_CONF_ENABLED 0` in it's project-conf.h.
9. Flash LP with web-demo and perform testing.
Attachments:
[6lbrQ.zip](https://github.com/contiki-os/contiki/files/747587/6lbrQ.zip)

jbartlett12026, 2017-02-02:
@vad32m This is very helpful. Thank you!