contiki-os/contiki #2016: CC26xx-web-demo unable to see sensorTag data on IBM quickstart

warisara, 2016-12-22:
I am following the procedure for the CC26xx web demo listed at http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples.
I can open 6lbr web page ([bbbb::100]) and can open web page on cc2650.
 
[image]

[image]

but in IBM quickstart I can't see sensorTad data.

[image]

I try to sniff packet on BBB by use command '"sudo tcpdump -i eth0 -v -X port 1883". it show below.

[image]

but I still can't see data in IBM quickstart.
Can anyone know how to solve this problem? please.

//Warisara

jonnteolsson, 2016-12-23:
The problem is in your NAT64 setup most probably. Using tcpdump -i eth0 -v -X port 1883 you should also see some IPv4 data, not only IPv6. Do you have the prefix for NAT64 correctly configured?

warisara, 2016-12-23:
@jonnteolsson 

My wrapsix configuration is

/* +++ CONFIGURATION +++ */
#define INTERFACE       "eth0"
#define PREFIX          "64:ff9b::"
#define IPV4_ADDR       "192.168.1.3"
#define HOST_IPV6_ADDR  "bbbb::101"
#define HOST_IPV4_ADDR  "192.168.1.42"
/* --- CONFIGURATION --- */

192.168.1.42 is Beaglebones IPv4 address. this configure is correct?

//warisara

chenek, 2017-01-06:
Actually, you don't have to run wrapsix if you run latest 6lbr which enable NAT64. You can follow the settings in the screen shot and do remember to set Peer router to IPv6 address of your desktop.

[image]

closed 2017-01-15 by g-oikonomou