contiki-os/contiki #2021: CoAP Reverse Proxy for Contiki

tarakanov, 2016-12-28:
Hi!
I am new to Contiki and interested in CoAP Reverse Proxy (as mentioned in RFC7252). I have a setup with 6lbr-border router (cc2538) and very sleepy sensor (sensortag). I would like to cache responses from the sensor in the border router. For this purpose I would like to use CoAP Reverse Proxy mechanism. But it seems that erbium server/client doesn't implement it. Am I correct? Is there any CoAP server with reverse proxy for Contiki by this time?

g-oikonomou, 2016-12-28:
Semyon we don't have that at the moment.

If you have a CoAP-aware IoT platform you can use it for your purpose and then 6lbr can simply route if you see what I mean.

nginxsantos, 2017-01-04:
Is there any CoAP based reverse proxy  open source solution available?  The popular ones like HaProxy and Nginx does not support CoAP.

closed 2017-01-10 by g-oikonomou