contiki-os/contiki #2030: CC1310 web-demo ack timeout

INemesisI, 2017-01-03:
Hello,
while trying to get the web-demo work on my CC1310s with 6lbr installed on a Raspberry PI, I noticed, that the website of the node gets slower and slower over time. After a look into the logs I discovered this:

> 2017-01-03 12:12:49.376655: ERROR: BR-RDC: br-rdc: ack received for unknown packet (14)
> 2017-01-03 12:12:49.571554: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (15)
> 2017-01-03 12:12:49.576573: ERROR: BR-RDC: br-rdc: ack received for unknown packet (15)
> 2017-01-03 12:12:49.770972: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (0)
> 2017-01-03 12:12:49.777107: ERROR: BR-RDC: br-rdc: ack received for unknown packet (0)
> 2017-01-03 12:12:49.971020: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (1)
> 2017-01-03 12:12:49.976648: ERROR: BR-RDC: br-rdc: ack received for unknown packet (1)
> 2017-01-03 12:12:50.171493: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (2)
> 2017-01-03 12:12:50.176627: ERROR: BR-RDC: br-rdc: ack received for unknown packet (2)
> 2017-01-03 12:12:50.371837: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (3)
> 2017-01-03 12:12:50.377032: ERROR: BR-RDC: br-rdc: ack received for unknown packet (3)
> 2017-01-03 12:12:50.572279: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (4)
> 2017-01-03 12:12:50.577482: ERROR: BR-RDC: br-rdc: ack received for unknown packet (4)
> 2017-01-03 12:19:08.597917: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (0)
> 2017-01-03 12:19:08.602127: ERROR: BR-RDC: br-rdc: ack received for unknown packet (0)
> 2017-01-03 12:19:08.798764: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (1)
> 2017-01-03 12:19:08.802015: ERROR: BR-RDC: br-rdc: ack received for unknown packet (1)
> 2017-01-03 12:19:08.999520: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (2)
> 2017-01-03 12:19:09.002765: ERROR: BR-RDC: br-rdc: ack received for unknown packet (2)
> 2017-01-03 12:19:09.199235: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (3)
> 2017-01-03 12:19:09.203559: ERROR: BR-RDC: br-rdc: ack received for unknown packet (3)
> 2017-01-03 12:19:09.399447: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (4)
> 2017-01-03 12:19:09.403304: ERROR: BR-RDC: br-rdc: ack received for unknown packet (4)
> 2017-01-03 12:19:09.599138: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (5)
> 2017-01-03 12:19:09.603477: ERROR: BR-RDC: br-rdc: ack received for unknown packet (5)
> 2017-01-03 12:19:09.799346: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (6)
> 2017-01-03 12:19:09.803240: ERROR: BR-RDC: br-rdc: ack received for unknown packet (6)
> 2017-01-03 12:19:09.999586: ERROR: BR-RDC: br-rdc: send failed, slip ack timeout (7)
> 2017-01-03 12:19:10.003448: ERROR: BR-RDC: br-rdc: ack received for unknown packet (7)

It surprises me a bit, that the unknown packet error happens always about 5ms after the timeout. Both nodes, the slip and the web-demo are physically placed next to each other.

BaN-Co, 2017-01-03:
I've fixed this by increasing the SIXLBR_CONFIG_DEFAULT_SLIP_TIMEOUT definition in the 6lbr/examples/6lbr/platform/native/native-config.h file from (CLOCK_SECOND / 5) to CLOCK_SECOND. I think it may be because the sub-hz CC1310 baudrate is 50kbps instead of the default 250kbps for the 2.4GHz devices, but I may be mistaken as I didn't dig deeper into this problem.

INemesisI, 2017-01-04:
Thank you very much, that fixed the problem!

closed 2017-01-04 by INemesisI