contiki-os/contiki #2031: RPL-ICMP6 uninitialized pointer

tozon, 2017-01-04:
dao_input_storing() function uses "rep" local variable.
It is initialized in L761.
It is used in multiple places (also in L854).
If RPL_WITH_MULTICAST=1 goto jump may happen from L757 to L841. In such case rep variable is used without initialization. This can lead to random behaviour or hard fault.

fprwi6labs, 2017-02-07:
Same bug encountered on branch release-3-1.
The multicast is not operational anymore.

yatch, 2017-02-07:
@tozon @fprwi6labs Could you try the patch of #2101 with your code to see the issue is resolved?

tozon, 2017-04-02:
@yatch I have verified the change. If DAO with multicast prefix is received, it is processed without any trouble now.
I see different problem now: DAO containing unicast prefix is resent in case of no ACK received. This mechanism is not implemented for DAO with multicast prefix and they often get lost. But this is other issue.

yatch, 2017-04-03:
@tozon Thank you for your test. I guess you're right. I'll send a PR as I mentioned in https://github.com/contiki-os/contiki/pull/2101#issuecomment-285236893 and we can discuss there :-)

closed 2017-04-18 by nvt