contiki-os/contiki #2036: Content-Format: 1541 problem with Contiki LWM2M and the latest Leshan releases

drugo72, 2017-01-07:
As of Leshan 0.1.11-M12 (M11 is the last one that works), temperature observations reported by the ``ipso-objects`` example are not properly decoded by Leshan that throws a null pointer exception.

The Contiki LWM2M implementation sends observations with ``Content-Format`` set to 1541 which is no longer supported/valid with the latest Leshan releases ([see this issue](https://github.com/eclipse/leshan/issues/155))

It seems that this can easily fixed by changing the default content format in a few places of ``lwm2m.engine.c`` from ``LWM2M_TEXT_PLAIN`` to just ``TEXT_PLAIN``.

This works however I'm not sure this is the proper fix.

joakimeriksson, 2017-01-10:
Yes, that is more or less it, but we also need to be backwards compliant. We are working on a significant update of LWM2M and how it is integrated with CoAP. In this I recently added support for the new types. It is currently at a temporary repository at github.com:nfi/contiki-lwm2m.git .We aim to move it to a more official repository soon and upstream changes when we have a robust implementation. I am also working on a port to NodeMCU for the complete LWM2M engine from Contiki using the new codebase.

drugo72, 2017-01-11:
Thanks. I'll give it a try.

closed 2017-01-11 by drugo72

joakimeriksson, 2017-01-11:
Ok, if you have any questions just send me an email!