contiki-os/contiki #2038: CC2650 flash

dominionish, 2017-01-09:
Hello!
I need to write some data to flash and have some trouble.
Every hour i write data to flash. Total: 24 records.
so i decided to use contiki file system, but it doesnt work:
[image]
Ok, then i tried to use cfs native but had the same problem.
Then I found example - cc26xx-web-demo, where struct saved directly to flash, using ext_flash_write.
The problem is - when I wrote data first time - it's all right - all data saved. When i tried to rewrite data - (the same offset, the same length) - data random transformed (wrote 13 value - but the written value was 2 or some other).
The I found ext_flash_erase function and before write data I erase it. And there a have a problem too.
Doesnt matter what offset or length set to ext_flash_erase - ALL flash will be erase. for example I has Values in Flash after first time:
23(offset - 0, length - 4) 34(offset-4, length - 4) 45(offset - 12, length - 4) 34(offset - 16, length - 4)
If I'll try to erase data on any offset and length = 4 Value will be:
-1 -1 -1 -1.
So, the questions are:
1. Why coffe - cfs and cfs native not working on cc2650
2. Why erasing small part of flash - erase ALL flash?
3. How to use ext_flash_erase to erase a part of flash?

g-oikonomou, 2017-01-10:
I am assuming that you are using one of the supported devices (Launchpad or Sensortag), and not some custom board.

> The problem is - when I wrote data first time - it's all right - all data saved. When i tried to rewrite data - (the same offset, the same length) - data random transformed (wrote 13 value - but the written value was 2 or some other).

You can't do that. With those flashes, you can write from 1 to zero, but not the other way round. If you observe the state before and after your 2nd write attempt, you will realise that you are not getting random data. You are just getting more 1s converted to 0s, depending on the value you are trying to write.

In order to write from 0 to 1 you basically need to erase first.

> If I'll try to erase data on any offset and length = 4 Value will be:
> -1 -1 -1 -1.

Perfectly normal. Erasing flash sets all bits to 1 (so you can then write as zero). Therefore all bytes to 0xFF (== your -1 if read as a signed integer)

> Why coffee - cfs and cfs native not working on cc2650

CFS needs platform-specific code in order to work for different platforms. We don't have that for CC2650.

> Why erasing small part of flash - erase ALL flash?

I need more detail here. Generally, you can only erase flash in sectors (of size 4096 for this particular flash). So, are you actually sure the entire flash is being erased in all cases? If so, then you may be looking at a bug.

> How to use ext_flash_erase to erase a part of flash?

https://github.com/contiki-os/contiki/blob/master/platform/srf06-cc26xx/common/ext-flash.h#L73

dominionish, 2017-01-17:
Thank you for your help!

closed 2017-01-17 by g-oikonomou

vitarico, 2017-01-28:
Hi!
@dominionish I am working on a similar problem. How do you read the flash? 
the ext_flash_read() function only returns a boolean value...

Thank you very much! 

 Victoria