contiki-os/contiki #2039: RF frontend interface

e120guru, 2017-01-10:
Don't know if this is the right place to post this, or if there already is something and I have just missed it... But, here we go:

We're working with a platform that in all essence is a cc2535 with a radio frontend (LNA and PA).

It would seem to me that it would be nice to have a layer far down below the radio driver that can use a frontend. It needs the following functions, and perhaps some more:

* Off (standby)
* RX
* TX
* Bypass
* Get TX power offset
* Get RX power offset

So basically the radio driver would put the frontend in RX or TX mode depending on what mode it is in itself. But since the frontend will affect output powers and RSSI readings, the module should be able to report it's offsets back to the radio driver so that the radio driver can use them for correcting the values it uses.

Platforms that does not have a frontend (like cc2538dk) could deklare the use of nullfrontend or something like that.

We're happy to implement this, but at least we would need to know that we do it in a way that is OK with the community so that we later on could create a pull request for it.

joakimeriksson, 2017-01-10:
Yes, sounds like a good thing to have. We have been working a bit with CC2538 and CC2592 and I guess we would have good use for such a front-end low-level interface!

alignan, 2017-01-10:
Sounds nice! I have a CC2538 + CC1200 (Zoul/RE-Mote) with PA for both RF interfaces so it would be definitely useful