contiki-os/contiki #2042: cc26xx-web-demo : Change MQTT configuration failed

warisara, 2017-01-11:
I am following the procedure for the CC26xx web demo listed at http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples.
then I can see data on IBM quickstart. 

[image]

I want to create app on IBM platform by use data on IBM quickstart 
I try to change  Org ID from quickstart to my Org ID and add Auth Token then device disconnected on IBM quickstart but in my app it's not connect too.

[image]

what should I do?

Warisara

g-oikonomou, 2017-01-11:
I vaguely remember reading something about Org IDs on Watson, somewhere. Can you check watson docs, will it still accept Org ID + Token in unencrypted MQTT messages?

warisara, 2017-01-13:
@g-oikonomou 

 when I try tutorials on IBM watson platform it have to use this information :
- tcp://<org-id>.messaging.internetofthings.ibmclound.com
- port : 1883
- org
- device type
- device id
- token

but in MQTT/IBM Cloud Config web it's difference.

[image]

Warisara

warisara, 2017-01-13:
@g-oikonomou 

Maybe it can't do?

[image]

g-oikonomou, 2017-01-15:
I think you may have to first register the device with Watson.

https://console.ng.bluemix.net/docs/services/IoT/devices/mqtt.html

g-oikonomou, 2017-02-18:
Any updates on that one?

closed 2017-03-07 by g-oikonomou

MarCast, 2017-10-06:
Hi, I working with CC13xx with MQTT. I open IBM Watson but I not read the value. Can you help me?