contiki-os/contiki #2068: control wsns with SDN controller

mokli, 2017-01-14:
Hi everyone,
I am newly in contiki. I want to connect an SDN controller (such as ONOS) to a wsn network and control my wsn with the SDN controller can any one help me to establish this connection?

BigMikes, 2017-01-14:
You could do that with the RPL-border-router example. Try to follow this guide: http://anrg.usc.edu/contiki/index.php/RPL_Border_Router

But be careful, ONOS assumes to talk using OpenFlow protocol, which must be implemented inside the sensor node.

mokli, 2017-01-16:
Thank you dear, can you tell me how can I implement open flow protocol in WSN?

alignan, 2017-01-16:
Use the Contiki mailing list to ask for help, the issue tracker is to report problems

closed 2017-01-16 by alignan

amirynia, 2019-06-03:
> 
> 
> Thank you dear, can you tell me how can I implement open flow protocol in WSN?

Did you find the solution?