contiki-os/contiki #2070: pinging nodes (more than one hop) with web-server example

rifotu, 2017-01-15:
Hello,

I have setup the below system

6LBR (1.4.1) with a CC1310 Launchpad slip radio
2x CC1350 ST with web-demo firmware 
2x CC1310 LP with web-demo firmware 

[image]

I can ping nodes / access their web pages that are 1 hop away from border router
But I can't ping node _d2F7 nor access its webpage?

> ping6 fd00::212:4b00:bcd:d2F7

I read a comment of @g-oikonomou  in an issue discussion, is this valid for my case too?

> You'll never be able to ping fe80::<anything> when it's more than one hop away (which in your case it is). I can't see anything wrong with the tunslip output. What happens if you change both BR and mesh node to use NullRDC instead?

How can I see that I can access to node _d2F7?  Should I use coap plugin?
Is there some documentation that I can read so that I can understand why I can't ping nodes more than one hop away?

thank you

g-oikonomou, 2017-01-15:
You should be able to ping :d2F7 without problems if you use its global IPv6 address (fd00::<something>) and not its link-local address (fe80::<something>)

rifotu, 2017-01-15:
Hello,

I will try pinging tomorrow again but with the below command (using fd00::) I wasn't able to ping d2F7 
> ping6 fd00::212:4b00:bcd:d2F7

Both _E5D6 and _D2F7 are CC1310 launchpads, do you think it would help if I switch to nullrdc?

g-oikonomou, 2017-01-15:
Switching to nullrdc should decrease delay and loss, but other than that the device should be reachable with ContikiMAC too.

rifotu, 2017-01-16:
Hello,

I switched to nullrdc and pinging works without problems. Later on I will try the contikimac again and write here a comparison.
thanks for your help :)

fzy-Line, 2017-03-14:
Could you tell me which web-demo do you run in nodes? When I run the 6lbr-demo in nodes, I could make my nodes connect with the 6lbr. But if I use the demo in the /contiki/examples, nodes couldn't connect with the 6lbr.