contiki-os/contiki #2081: Contiki IDE Configuration

msaiducar, 2017-01-19:
Hi everyone,

I'm trying to setup and development environment for contiki but I couldn't succeed yet. I found one tutorial about settings [1] but it didn't work. I can build example projects with command line on both Ubuntu and Windows but it is a little bit slow and easy to make mistake. What I want is import contiki project to Eclipse (or any other IDE) and develop application and build in IDE itself. I import contiki source to Eclipse but there is a lot of errors ("Type uint8_t could not resolved etc). 

I appreciate of any suggestion,
Best Regards,

Said.  

[1] = https://github.com/Zolertia/Resources/wiki/Eclipse

vad32m, 2017-01-20:
I am not sure, that syntax highlighting should work, as for me, I succeed with connecting my CC1310 project to the CCS6(which is Eclipse-based solution), but syntax highlighter works pretty bad. Navigation through code is not perfect too. I think this may be caused by using external makefile, which is only way to build Contiki. So IDE cant perform just in time compilation. Anyway, you still can perform debugging from Eclipse through JTAG(at least on my platform), which is much better, then playing with GDB console.

thomas-blank, 2017-02-01:
What works for me with proper highlighting, code navigation, etc, using Eclipse Neon and installed GNU ARM Eclipse 3.2.1:
1. New Project > Makefile Project with Existing Code
2. Name: <Whatever>, Code Location <your contiki location>. Languages: C. Toolchain: Cross ARM GCC Toolchain
3. Add a new build Configuration for every binary that you want to build, e.g. examples/ipv6/rpl-border-router by going to Project > Properties > C/C++ Build > Manage Configurations. Set it Active.
4. Still on Project > Properties > C/C++ Build, on the Builder Settings Tab: Unset Use default build command, set Build Command to "make", unset Generate Makefiles automatically, set Build Location to where the Makefile resides, so for this example: `${workspace_loc:/contiki/examples/ipv6/rpl-border-router}` (Assuming you used 'contiki' as a project name.)
5. On C/C++ Build > Environment, add Variables V=1 and TARGET=<target platform>.
6. On C/C++ General > Preprocessor Include Paths, Macros etc. > Providers Tab, enable CDT GCC Built-In Compiler Settings Cross ARM and CDT GCC Build Output Parser. In the Provider Options for the Output Parser, add "|(arm-none-eabi-gcc)" to the Compiler command pattern and set Container to keep discovered entries to Project. In the Options for the Built-In Compiler Settings, in the Command to get compiler specs field, replace ${COMMAND} with arm-none-eabi-gcc.
7. ON C/C++ General > Indexer, Enable project specific settings. Set Build Configuration for the indexer to Use active build configuration.

For every new binary, add a new Build configuration that you base off of the old one. All settings will be transferred. Adapt the build path to the new Makefile, and the TARGET to your target.

Clean and rebuild. Everything should now work as expected.