contiki-os/contiki #2095: CC1310: 'rpl-tsch' example question

morphal, 2017-02-02:
Hi!
I am trying to get into TSCH technology with contiki. I decided to start with rpl-tsch example and from very start I faced difficulties and need some help.
**CC1310 Launchpad:**
1. `make TARGET=srf06-cc26xx BOARD=launchpad/cc1310` in '_examples/ipv6/rpl-tsch_' folder
2. After flashing and connecting to the launchpad via serial I see this log:
```
Starting Contiki-3.x-3153-g263af6f
With DriverLib v0.47020
TI CC1310 LaunchPad
IEEE 802.15.4: No, Sub-GHz: Yes, BLE: No, Prop: No
 Net: sicslowpan
 MAC: TSCH
 RDC: nordc
TSCH:! radio does not support getting RADIO_PARAM_RX_MODE. Abort init.
 RF: Channel 0
 Node ID: 53254
Init: current role: 6ln. Will start in 5 seconds. Press user button to toggle mo                                                                                                                                                             de.
Init: node starting with role 6ln
--- Network status ---
- Server IPv6 addresses:
-- fe80::212:4b00:e07:d006
- Default route:
-- None
- Routing links (0 in total):
----------------------
TSCH:! not initialized (see earlier logs), drop outgoing packet
...
```
Putting prop-mode.c in DEBUG mode didnt show any additional info about the reasons, why the error does appear:
`TSCH:! radio does not support getting RADIO_PARAM_RX_MODE. Abort init.`

Can anyone help me to get it working, please?

simonduq, 2017-02-02:
TSCH is not ported to cc1310 yet, unfortunately. See https://github.com/contiki-os/contiki/tree/master/core/net/mac/tsch#porting-tsch-to-a-new-platform
for instructions how do to such a port. If you do so, please consider doing a pull request!

closed 2017-02-02 by simonduq

morphal, 2017-02-02:
@simonduq, can you correct me if I am wrong?
As I understand CC1310 shares platform `srf06-cc26xx`, which is mentioned as already supported (as well as CC2650 cpu). Does this mean that the port is done not for the platform, but for the specific cpu?

simonduq, 2017-02-02:
Right, the `srf06-cc26x` port currently only supports the cc2650 2.4 GH RF module, not cc1310.