contiki-os/contiki #2104: Web browser on Smart phone for 6lowpan of cc1350

DonGody, 2017-02-09:
Hi Everyone
I have been testing 6lowpan with edge router based raspberry pi3 and sensor nodes based cc1350 launchpad. 
I have succeeded to access to sensor node web page and CoAP web page on Firefox web browser of Windows OS and Ubuntu OS.
But, I have problems on web browser of smart phone.
I have succeeded to access to 6lbr first page and sensor list page on web browser of smart phone.
But, I have failed to access to sensor node page and CoAP page on it.
I am sure that it is the problem related to setting of web browser of smart phone.
Can you support me?

g-oikonomou, 2017-02-18:
I've never tested this functionality with a browser on a smartphone. I am not even sure if those browsers are happy with IPv6 addresses in the address bar.

chenek, 2017-02-20:
According to my test, it's also OK to type in [bbbb::100] in chrome browser and see 6lbr web page on Android. However, it failed to access to sensor node page and CoAP page.

casaroli, 2017-02-20:
I had some problems with TCP in the past because of buffer configuration in
rpl-border-router

I just removed the stuff from project-conf.h so that it uses the platform
defaults and it worked fine with phones and other browsers.

May I suggest you try the same?

BR

chenek, 2017-02-20:
Can you specify which project-confirm.h and what I should remove to test again?

casaroli, 2017-02-20:
If you are using slip-radio, check examples/ipv6/slip-radio/project-conf.h

If you are using rpl-border-router, check
examples/ipv6/rpl-border-router/project-conf.h

Remove or comment out the following lines:

#ifndef QUEUEBUF_CONF_NUM
#define QUEUEBUF_CONF_NUM          4
#endif

#ifndef UIP_CONF_BUFFER_SIZE
#define UIP_CONF_BUFFER_SIZE    140
#endif

#ifndef UIP_CONF_RECEIVE_WINDOW
#define UIP_CONF_RECEIVE_WINDOW  60
#endif

Marco Casaroli

chenek, 2017-02-21:
Just test with you suggestion but it still doesn't work.

casaroli, 2017-02-21:
Hey YK, can you check tcpdump (or wireshark) and see what is going on that
interface?

Marco Casaroli

chenek, 2017-02-21:
Shall I tcpdump port 80?

chenek, 2017-02-22:
@casaroli  I use tcpdump to sniff port 80 and see the following output when I access to sensors.html in 6lbr. But there is nothing output when I click on the "web" link on sensors page and my Android Chrome browser shows cannot connect to this web site.

sudo tcpdump -X -v -i eth0 port 80
tcpdump: listening on eth0, link-type EN10MB (Ethernet), capture size 65535 bytes
01:11:39.172151 IP6 (hlim 64, next-header TCP (6) payload length: 40) bbbb::4058:23d:fb03:a59b.49083 > bbbb::100.http: Flags [S], cksum 0x3b1a (correct), seq 390564082, win 14400, options [mss 1440,sackOK,TS val 137416 ecr 0,nop,wscale 6], length 0
        0x0000:  6000 0000 0028 0640 bbbb 0000 0000 0000  `....(.@........
        0x0010:  4058 023d fb03 a59b bbbb 0000 0000 0000  @X.=............
        0x0020:  0000 0000 0000 0100 bfbb 0050 1747 88f2  ...........P.G..
        0x0030:  0000 0000 a002 3840 3b1a 0000 0204 05a0  ......8@;.......
        0x0040:  0402 080a 0002 18c8 0000 0000 0103 0306  ................
01:11:39.172474 IP6 (hlim 64, next-header TCP (6) payload length: 24) bbbb::100.http > bbbb::4058:23d:fb03:a59b.49083: Flags [S.], cksum 0xda16 (correct), seq 718, ack 390564083, win 48, options [mss 1220], length 0
        0x0000:  6000 0000 0018 0640 bbbb 0000 0000 0000  `......@........
        0x0010:  0000 0000 0000 0100 bbbb 0000 0000 0000  ................
        0x0020:  4058 023d fb03 a59b 0050 bfbb 0000 02ce  @X.=.....P......
        0x0030:  1747 88f3 6012 0030 da16 0000 0204 04c4  .G..`..0........
01:11:39.175315 IP6 (hlim 64, next-header TCP (6) payload length: 20) bbbb::4058:23d:fb03:a59b.49083 > bbbb::100.http: Flags [.], cksum 0xb8d3 (correct), ack 1, win 14400, length 0
        0x0000:  6000 0000 0014 0640 bbbb 0000 0000 0000  `......@........
        0x0010:  4058 023d fb03 a59b bbbb 0000 0000 0000  @X.=............
        0x0020:  0000 0000 0000 0100 bfbb 0050 1747 88f3  ...........P.G..
        0x0030:  0000 02cf 5010 3840 b8d3 0000            ....P.8@....
01:11:39.183559 IP6 (hlim 64, next-header TCP (6) payload length: 68) bbbb::4058:23d:fb03:a59b.49083 > bbbb::100.http: Flags [P.], cksum 0xa4b5 (correct), seq 1:49, ack 1, win 14400, length 48
        0x0000:  6000 0000 0044 0640 bbbb 0000 0000 0000  `....D.@........
        0x0010:  4058 023d fb03 a59b bbbb 0000 0000 0000  @X.=............
        0x0020:  0000 0000 0000 0100 bfbb 0050 1747 88f3  ...........P.G..
        0x0030:  0000 02cf 5018 3840 a4b5 0000 4745 5420  ....P.8@....GET.
        0x0040:  2f73 656e 736f 7273 2e68 746d 6c20 4854  /sensors.html.HT
        0x0050:  5450 2f31 2e31 0d0a 486f 7374 3a20 5b62  TP/1.1..Host:.[b
        0x0060:  6262 623a 3a31 3030 5d0d 0a43            bbb::100]..C
01:11:39.183798 IP6 (hlim 64, next-header TCP (6) payload length: 105) bbbb::100.http > bbbb::4058:23d:fb03:a59b.49083: Flags [P.], cksum 0x5f27 (correct), seq 1:86, ack 49, win 48, length 85
        0x0000:  6000 0000 0069 0640 bbbb 0000 0000 0000  `....i.@........
        0x0010:  0000 0000 0000 0100 bbbb 0000 0000 0000  ................
        0x0020:  4058 023d fb03 a59b 0050 bfbb 0000 02cf  @X.=.....P......
        0x0030:  1747 8923 5018 0030 5f27 0000 4854 5450  .G.#P..0_'..HTTP
        0x0040:  2f31 2e30 2032 3030 204f 4b0d 0a53 6572  /1.0.200.OK..Ser
        0x0050:  7665 723a 2043 6f6e 7469 6b69 2f32 2e34  ver:.Contiki/2.4
        0x0060:  2068 7474 703a 2f2f 7777 772e 7369 6373  .http://www.sics
        0x0070:  2e73 652f 636f 6e74 696b 692f 0d0a 436f  .se/contiki/..Co
        0x0080:  6e6e 6563 7469 6f6e 3a20 636c 6f73 650d  nnection:.close.
        0x0090:  0a                                       .
01:11:39.188466 IP6 (hlim 64, next-header TCP (6) payload length: 68) bbbb::4058:23d:fb03:a59b.49083 > bbbb::100.http: Flags [P.], cksum 0x0e39 (correct), seq 49:97, ack 86, win 14400, length 48
        0x0000:  6000 0000 0044 0640 bbbb 0000 0000 0000  `....D.@........
        0x0010:  4058 023d fb03 a59b bbbb 0000 0000 0000  @X.=............
        0x0020:  0000 0000 0000 0100 bfbb 0050 1747 8923  ...........P.G.#
        0x0030:  0000 0324 5018 3840 0e39 0000 6f6e 6e65  ...$P.8@.9..onne
        0x0040:  6374 696f 6e3a 206b 6565 702d 616c 6976  ction:.keep-aliv
        0x0050:  650d 0a43 6163 6865 2d43 6f6e 7472 6f6c  e..Cache-Control
        0x0060:  3a20 6d61 782d 6167 653d 300d            :.max-age=0.
01:11:39.188664 IP6 (hlim 64, next-header TCP (6) payload length: 47) bbbb::100.http > bbbb::4058:23d:fb03:a59b.49083: Flags [P.], cksum 0x34c3 (correct), seq 86:113, ack 97, win 48, length 27
        0x0000:  6000 0000 002f 0640 bbbb 0000 0000 0000  `..../.@........
        0x0010:  0000 0000 0000 0100 bbbb 0000 0000 0000  ................
        0x0020:  4058 023d fb03 a59b 0050 bfbb 0000 0324  @X.=.....P.....$
        0x0030:  1747 8953 5018 0030 34c3 0000 436f 6e74  .G.SP..04...Cont
        0x0040:  656e 742d 7479 7065 3a20 7465 7874 2f68  ent-type:.text/h
        0x0050:  746d 6c0d 0a0d 0a                        tml....
01:11:39.193269 IP6 (hlim 64, next-header TCP (6) payload length: 68) bbbb::4058:23d:fb03:a59b.49083 > bbbb::100.http: Flags [P.], cksum 0x7d6d (correct), seq 97:145, ack 113, win 14400, length 48
        0x0000:  6000 0000 0044 0640 bbbb 0000 0000 0000  `....D.@........
        0x0010:  4058 023d fb03 a59b bbbb 0000 0000 0000  @X.=............
        0x0020:  0000 0000 0000 0100 bfbb 0050 1747 8953  ...........P.G.S
        0x0030:  0000 033f 5018 3840 7d6d 0000 0a55 7067  ...?P.8@}m...Upg
[... cut ...]
01:11:39.239344 IP6 (hlim 64, next-header TCP (6) payload length: 38) bbbb::4058:23d:fb03:a59b.49083 > bbbb::100.http: Flags [P.], cksum 0xa3b8 (correct), seq 481:499, ack 1786, win 18040, length 18
        0x0000:  6000 0000 0026 0640 bbbb 0000 0000 0000  `....&.@........
        0x0010:  4058 023d fb03 a59b bbbb 0000 0000 0000  @X.=............
        0x0020:  0000 0000 0000 0100 bfbb 0050 1747 8ad3  ...........P.G..
        0x0030:  0000 09c8 5018 4678 a3b8 0000 713d 302e  ....P.Fx....q=0.
        0x0040:  362c 656e 3b71 3d30 2e34 0d0a 0d0a       6,en;q=0.4....

neophob, 2017-02-22:
im not sure to understand this request correctly, but since when do browser support coap (without plugins?)

chenek, 2017-02-22:
@neophob I access to web page of sensor not COAP.

g-oikonomou, 2017-04-24:
Where are we on this one folks? Can we close?