contiki-os/contiki #2106: CC1310 Launchpad compilation error for slip-radio

electronut, 2017-02-09:
In contiki/examples/ipv6/slip-radio:

`make TARGET=srf06-cc26xx BOARD=launchpad/cc1310`

gives the following error:

```
contiki-srf06-cc26xx.a(srf06-sensors.o):(.data.sensors+0x0): undefined reference to `button_select_sensor'
contiki-srf06-cc26xx.a(srf06-sensors.o):(.data.sensors+0xc): undefined reference to `button_up_sensor'
contiki-srf06-cc26xx.a(srf06-sensors.o):(.data.sensors+0x10): undefined reference to `button_down_sensor'
collect2: error: ld returned 1 exit status
make: *** [slip-radio.elf] Error 1
```

Other examples like `rpl-border-router` are building fine for `launchpad/CC1310`.

g-oikonomou, 2017-02-09:
Try `make clean` and give it another go

electronut, 2017-02-09:
Boy do I feel foolish. Solved. Thanks. :-)

closed 2017-02-09 by electronut

rachana7, 2017-12-10:
electronut ,I get the same error,so could you please explain how you solved that error