contiki-os/contiki #2111: Can not turn of radio chip CC2630 on the latest contiki and CC26xxware driver

chitrung218ulsan, 2017-02-16:
Hi,

I am developing a protocol on chip CC2630 with my custom board. I configure two parameters in ccfg.c as follows since I do not use external capacitors for crystal 24Mhz
 #define SET_CCFG_MODE_CONF_XOSC_CAP_MOD              0x0        // Apply cap-array delta
#define SET_CCFG_MODE_CONF_XOSC_CAPARRAY_DELTA          0x00

When I use the latest contiki and CC26xxware driver, the radio chip cannot be turned on. The errors appeared as follows.

rf_cmd_ieee_rx: ret=0, CMDSTA=0x00000082, status=0x0000
init: on() failed

However, when I tried with the older version of CC26xxware, it worked. Therefore, could you help me find the solution why it does not work on the latest version of contiki and CC26xxware driver.

g-oikonomou, 2017-02-18:
This is most likely related to overrides. Which command to the radio core is it that fails? My gut feeling is that it is failure to sync the FS.

g-oikonomou, 2017-02-18:
I am also somewhat confused by your use of the terms "older version of cc26xxware". What version is it that works, what version is it that does not? #1931 might help.

chitrung218ulsan, 2017-02-19:
The latest version of CC26xxware is 2.24.03.17272. The older version is v2.23.02.16941.

The problem I suffered is that the radio cannot be turned on in ieee-mode.
The debug code is as follows.

In function init(void) in ieee-mode.c

`if((on) ! RF_CORE_CMD_OK)
{
PRINTF("init: on() failed \n");
return RF_CORE_CMD_ERROR;
}
`

Then, in function rf_is_on in ieee-mode.c, rf_core is not accessible.

Could you figure out the problem.

chitrung218ulsan, 2017-02-19:
> This is most likely related to overrides. Which command to the radio core is it that fails? My gut feeling is that it is failure to sync the FS.

How can I fixed this one?

phantomgz, 2017-08-04:
Hi g-oikonomou:
  Take a look of my pull request, I think I have fixed this issue. 
https://github.com/g-oikonomou/cc26xxware/pulls

Cai.