contiki-os/contiki #2114: Add support for board based on CC1310 4x4mm (RSM) package

henkyprayoga, 2017-02-18:
The platform currently support CC1310 7x7mm package. I've tried to run it on 4x4 but it locks the debugger interface. I've modified board.h but still doesn't work. Does anyone has similar experience? Here's the board.h modification. Thank you very much.

`/*
 * Copyright (c) 2016, Texas Instruments Incorporated - http://www.ti.com/
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 * 1. Redistributions of source code must retain the above copyright
 *    notice, this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *    notice, this list of conditions and the following disclaimer in the
 *    documentation and/or other materials provided with the distribution.
 * 3. Neither the name of the copyright holder nor the names of its
 *    contributors may be used to endorse or promote products derived
 *    from this software without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
 * ``AS IS'' AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 * LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
 * FOR A PARTICULAR PURPOSE ARE DISCLAIMED.  IN NO EVENT SHALL THE
 * COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
 * INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
 * (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
 * SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
 * HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT,
 * STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
 * ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED
 * OF THE POSSIBILITY OF SUCH DAMAGE.
 */

/*---------------------------------------------------------------------------*/
#ifndef BOARD_H_
#define BOARD_H_
/*---------------------------------------------------------------------------*/
#include "ioc.h"
/*---------------------------------------------------------------------------*/
/**
 * \name LED configurations
 *
 * Those values are not meant to be modified by the user
 * @{
 */
/* Some files include leds.h before us, so we need to get rid of defaults in
 * leds.h before we provide correct definitions */
#undef LEDS_GREEN
#undef LEDS_YELLOW
#undef LEDS_RED
#undef LEDS_CONF_ALL

#define LEDS_RED       1
#define LEDS_GREEN     2
#define LEDS_YELLOW    LEDS_GREEN
#define LEDS_ORANGE    LEDS_RED

#define LEDS_CONF_ALL  3

/* Notify various examples that we have LEDs */
#define PLATFORM_HAS_LEDS        1
/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \name LED IOID mappings
 *
 * Those values are not meant to be modified by the user
 * @{
 */
 
#define BOARD_IOID_LED_1          IOID_UNUSED
#define BOARD_IOID_LED_2          IOID_UNUSED
#define BOARD_LED_1               (1 << BOARD_IOID_LED_1)
#define BOARD_LED_2               (1 << BOARD_IOID_LED_2)
#define BOARD_LED_ALL             (BOARD_LED_1 | BOARD_LED_2)

/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \name UART IOID mapping
 *
 * Those values are not meant to be modified by the user
 * @{
 */
#define BOARD_IOID_UART_RX        IOID_1
#define BOARD_IOID_UART_TX        IOID_2
#define BOARD_IOID_UART_RTS       IOID_5
#define BOARD_IOID_UART_CTS       IOID_6
#define BOARD_UART_RX             (1 << BOARD_IOID_UART_RX)
#define BOARD_UART_TX             (1 << BOARD_IOID_UART_TX)
#define BOARD_UART_RTS            (1 << BOARD_IOID_UART_RTS)
#define BOARD_UART_CTS            (1 << BOARD_IOID_UART_CTS)
/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \name Button IOID mapping
 *
 * Those values are not meant to be modified by the user
 * @{
 */
#define BOARD_IOID_KEY_LEFT       IOID_7
#define BOARD_IOID_KEY_RIGHT      IOID_8
#define BOARD_KEY_LEFT            (1 << BOARD_IOID_KEY_LEFT)
#define BOARD_KEY_RIGHT           (1 << BOARD_IOID_KEY_RIGHT)
/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \brief SPI IOID mappings
 *
 * Those values are not meant to be modified by the user
 * @{
 */
#define BOARD_IOID_SPI_MOSI       IOID_UNUSED
#define BOARD_IOID_SPI_MISO       IOID_UNUSED
/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \name External flash IOID mapping
 *
 * Those values are not meant to be modified by the user
 * @{
 */
#define BOARD_IOID_FLASH_CS       IOID_UNUSED
#define BOARD_FLASH_CS            (1 << BOARD_IOID_FLASH_CS)
#define BOARD_IOID_SPI_CLK_FLASH  IOID_UNUSED
/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \brief I2C IOID mappings
 *
 * Those values are not meant to be modified by the user
 * @{
 */
#define BOARD_IOID_SCL            IOID_UNUSED
#define BOARD_IOID_SDA            IOID_UNUSED
/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \brief ROM bootloader configuration
 *
 * Change SET_CCFG_BL_CONFIG_BL_PIN_NUMBER to BOARD_IOID_KEY_xyz to select
 * which button triggers the bootloader on reset.
 *
 * The remaining values are not meant to be modified by the user
 * @{
 */
#if ROM_BOOTLOADER_ENABLE
#define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE            0xC5
#define SET_CCFG_BL_CONFIG_BL_LEVEL                     0x00
#define SET_CCFG_BL_CONFIG_BL_PIN_NUMBER                BOARD_IOID_KEY_LEFT
#define SET_CCFG_BL_CONFIG_BL_ENABLE                    0xC5
#else
#define SET_CCFG_BL_CONFIG_BOOTLOADER_ENABLE            0x00
#define SET_CCFG_BL_CONFIG_BL_LEVEL                     0x01
#define SET_CCFG_BL_CONFIG_BL_PIN_NUMBER                0xFF
#define SET_CCFG_BL_CONFIG_BL_ENABLE                    0xFF
#endif
/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \brief Remaining pins
 *
 * Those values are not meant to be modified by the user
 * @{
 */
//#define BOARD_IOID_CS             IOID_7
#define BOARD_IOID_TDO            IOID_3
#define BOARD_IOID_TDI            IOID_4
#define BOARD_IOID_DIO9           IOID_9

/** @} */
/*---------------------------------------------------------------------------*/
/**
 * \name Device string used on startup
 * @{
 */
#define BOARD_STRING "CC1310 4x4 RSM"

/** @} */
/*---------------------------------------------------------------------------*/
#endif /* BOARD_H_ */
/*---------------------------------------------------------------------------*/
/**
 * @}
 * @}
 */
`

g-oikonomou, 2017-02-18:
It's really difficult to understand what you are trying to ask here. What board did you use that had a 4x4 chip? Please provide a diff for any changes you made to any files rather than just pasting the entire file here. And please enclose your code in markdown code blocks.

henkyprayoga, 2017-02-18:
Hi,

First of all, thanks for your good intention to help. Ok, sorry. This is my first time posting on Github. We are building our own RF modules based on CC1310 (schematic attached). We have tested the module using SmartRF Studio 7 and it's work well. Here's the diff between changes.

```diff
--- platform/srf06-cc26xx/launchpad/cc1310/board.h
+++ platform/mcmv2/default/cc1310/board.modified.h
@@ -27,28 +27,8 @@
  * ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED
  * OF THE POSSIBILITY OF SUCH DAMAGE.
  */
+
 /*---------------------------------------------------------------------------*/
-/** \addtogroup launchpad-peripherals
- * @{
- *
- * \defgroup launchpad-cc1310-specific CC1310 LaunchPad Peripherals
- *
- * Defines related to the CC1310 LaunchPad
- *
- * This file provides connectivity information on LEDs, Buttons, UART and
- * other peripherals
- *
- * This file is not meant to be modified by the user.
- * @{
- *
- * \file
- * Header file with definitions related to the I/O connections on the TI
- * CC1310 LaunchPad
- *
- * \note   Do not include this file directly. It gets included by contiki-conf
- *         after all relevant directives have been set.
- */
-/*---------------------------------------------------------------------------*/
 #ifndef BOARD_H_
 #define BOARD_H_
 /*---------------------------------------------------------------------------*/
@@ -84,11 +64,13 @@
  * Those values are not meant to be modified by the user
  * @{
  */
-#define BOARD_IOID_LED_1          IOID_6
-#define BOARD_IOID_LED_2          IOID_7
+ 
+#define BOARD_IOID_LED_1          IOID_UNUSED
+#define BOARD_IOID_LED_2          IOID_UNUSED
 #define BOARD_LED_1               (1 << BOARD_IOID_LED_1)
 #define BOARD_LED_2               (1 << BOARD_IOID_LED_2)
 #define BOARD_LED_ALL             (BOARD_LED_1 | BOARD_LED_2)
+
 /** @} */
 /*---------------------------------------------------------------------------*/
 /**
@@ -97,10 +79,10 @@
  * Those values are not meant to be modified by the user
  * @{
  */
-#define BOARD_IOID_UART_RX        IOID_2
-#define BOARD_IOID_UART_TX        IOID_3
-#define BOARD_IOID_UART_RTS       IOID_18
-#define BOARD_IOID_UART_CTS       IOID_19
+#define BOARD_IOID_UART_RX        IOID_1
+#define BOARD_IOID_UART_TX        IOID_2
+#define BOARD_IOID_UART_RTS       IOID_5
+#define BOARD_IOID_UART_CTS       IOID_6
 #define BOARD_UART_RX             (1 << BOARD_IOID_UART_RX)
 #define BOARD_UART_TX             (1 << BOARD_IOID_UART_TX)
 #define BOARD_UART_RTS            (1 << BOARD_IOID_UART_RTS)
@@ -113,8 +95,8 @@
  * Those values are not meant to be modified by the user
  * @{
  */
-#define BOARD_IOID_KEY_LEFT       IOID_13
-#define BOARD_IOID_KEY_RIGHT      IOID_14
+#define BOARD_IOID_KEY_LEFT       IOID_7
+#define BOARD_IOID_KEY_RIGHT      IOID_8
 #define BOARD_KEY_LEFT            (1 << BOARD_IOID_KEY_LEFT)
 #define BOARD_KEY_RIGHT           (1 << BOARD_IOID_KEY_RIGHT)
 /** @} */
@@ -125,8 +107,8 @@
  * Those values are not meant to be modified by the user
  * @{
  */
-#define BOARD_IOID_SPI_MOSI       IOID_9
-#define BOARD_IOID_SPI_MISO       IOID_8
+#define BOARD_IOID_SPI_MOSI       IOID_UNUSED
+#define BOARD_IOID_SPI_MISO       IOID_UNUSED
 /** @} */
 /*---------------------------------------------------------------------------*/
 /**
@@ -135,9 +117,9 @@
  * Those values are not meant to be modified by the user
  * @{
  */
-#define BOARD_IOID_FLASH_CS       IOID_20
+#define BOARD_IOID_FLASH_CS       IOID_UNUSED
 #define BOARD_FLASH_CS            (1 << BOARD_IOID_FLASH_CS)
-#define BOARD_IOID_SPI_CLK_FLASH  IOID_10
+#define BOARD_IOID_SPI_CLK_FLASH  IOID_UNUSED
 /** @} */
 /*---------------------------------------------------------------------------*/
 /**
@@ -146,8 +128,8 @@
  * Those values are not meant to be modified by the user
  * @{
  */
-#define BOARD_IOID_SCL            IOID_4
-#define BOARD_IOID_SDA            IOID_5
+#define BOARD_IOID_SCL            IOID_UNUSED
+#define BOARD_IOID_SDA            IOID_UNUSED
 /** @} */
 /*---------------------------------------------------------------------------*/
 /**
@@ -178,28 +160,18 @@
  * Those values are not meant to be modified by the user
  * @{
  */
-#define BOARD_IOID_CS             IOID_11
-#define BOARD_IOID_TDO            IOID_16
-#define BOARD_IOID_TDI            IOID_17
-#define BOARD_IOID_DIO12          IOID_12
-#define BOARD_IOID_DIO15          IOID_15
-#define BOARD_IOID_DIO21          IOID_21
-#define BOARD_IOID_DIO22          IOID_22
-#define BOARD_IOID_DIO23          IOID_23
-#define BOARD_IOID_DIO24          IOID_24
-#define BOARD_IOID_DIO25          IOID_25
-#define BOARD_IOID_DIO26          IOID_26
-#define BOARD_IOID_DIO27          IOID_27
-#define BOARD_IOID_DIO28          IOID_28
-#define BOARD_IOID_DIO29          IOID_29
-#define BOARD_IOID_DIO30          IOID_30
+//#define BOARD_IOID_CS             IOID_7
+#define BOARD_IOID_TDO            IOID_3
+#define BOARD_IOID_TDI            IOID_4
+#define BOARD_IOID_DIO9           IOID_9
+
 /** @} */
 /*---------------------------------------------------------------------------*/
 /**
  * \name Device string used on startup
  * @{
  */
-#define BOARD_STRING "TI CC1310 LaunchPad"
+#define BOARD_STRING "CC1310 4x4 RSM"
 
 /** @} */
 /*---------------------------------------------------------------------------*/
```
Thanks!

g-oikonomou, 2017-02-18:
I cannot immediately see anything obviously wrong here. You _are_ using the 128KB flash chip, right? #1831 should help if you are using a chip with a smaller flash.

henkyprayoga, 2017-02-18:
Yes, we are using 128KB flash. I'm so confused with this problem. Contiki works perfectly on the Launchpad CC1310.  But once I flashed into our custom module, it doesn't work. The chip back to live after we perform mass erase though.

g-oikonomou, 2017-02-18:
Any chance the launchpad code is trying to initialisesomething that does not exist on your board? I see you have defined SPI and flash DIOs as unused, have you tried explicitly excluding the external flash functionality from the build?

Check the code in `board.c`, it tries to configure unused pins to a low-leakage state. Perhaps something dodgy happening there that you need to change.

g-oikonomou, 2017-02-18:
Suggest you also try merging #1932 into your local branch. There are some nice updates to the CC13xx port there.

henkyprayoga, 2017-02-18:
Probably... I'm still working on that... I also tried to compare it with CC2650EM-4XS. TI produced a reference design and examples for CC2650 4x4mm package. I've compared the pin out, its totally the same. 

Thanks for the hint. I will post back here soon after I got something.

g-oikonomou, 2017-02-18:
I should say that, even though our "officially" supported boards all use 7x7, it would be great to be able to add 4x4 and 5x5 packages to our list of supported features.

henkyprayoga, 2017-02-18:
Agree! Let's see if what I'm currently doing here could give small contribution...

henkyprayoga, 2017-02-20:
I have an update. Turns out that debugger locking also occured on 7x7. So the debugger lock out is another problem. And looks like it's expected behaviour. I've updated the startup_files/ccfg.c on cc13xxware subdirectory and the locking doesn't happen again.

```diff
diff --git a/startup_files/ccfg.c b/startup_files/ccfg.c
index 92fb448..08bfa60 100644
--- a/startup_files/ccfg.c
+++ b/startup_files/ccfg.c
@@ -228,8 +228,8 @@
 #endif

 #ifndef SET_CCFG_CCFG_TAP_DAP_0_CPU_DAP_ENABLE
-#define SET_CCFG_CCFG_TAP_DAP_0_CPU_DAP_ENABLE       0x00       // Access disabled
-// #define SET_CCFG_CCFG_TAP_DAP_0_CPU_DAP_ENABLE          0xC5       // Access enabled if also enabled in FCFG
+// #define SET_CCFG_CCFG_TAP_DAP_0_CPU_DAP_ENABLE       0x00       // Access disabled
+#define SET_CCFG_CCFG_TAP_DAP_0_CPU_DAP_ENABLE          0xC5       // Access enabled if also enabled in FCFG
 #endif

 #ifndef SET_CCFG_CCFG_TAP_DAP_0_PRCM_TAP_ENABLE
@@ -238,8 +238,8 @@
 #endif

 #ifndef SET_CCFG_CCFG_TAP_DAP_0_TEST_TAP_ENABLE
-#define SET_CCFG_CCFG_TAP_DAP_0_TEST_TAP_ENABLE      0x00       // Access disabled
-// #define SET_CCFG_CCFG_TAP_DAP_0_TEST_TAP_ENABLE         0xC5       // Access enabled if also enabled in FCFG
+// #define SET_CCFG_CCFG_TAP_DAP_0_TEST_TAP_ENABLE      0x00       // Access disabled
+#define SET_CCFG_CCFG_TAP_DAP_0_TEST_TAP_ENABLE         0xC5       // Access enabled if also enabled in FCFG
 #endif

 #ifndef SET_CCFG_CCFG_TAP_DAP_1_PBIST2_TAP_ENABLE
warning: LF will be replaced by CRLF in startup_files/ccfg.c.
The file will have its original line endings in your working directory.

D:\IoT\contiki\cpu\cc26xx-cc13xx\lib\cc13xxware>git diff
diff --git a/startup_files/ccfg.c b/startup_files/ccfg.c
index 92fb448..08bfa60 100644
--- a/startup_files/ccfg.c
+++ b/startup_files/ccfg.c
@@ -228,8 +228,8 @@
 #endif

 #ifndef SET_CCFG_CCFG_TAP_DAP_0_CPU_DAP_ENABLE
-#define SET_CCFG_CCFG_TAP_DAP_0_CPU_DAP_ENABLE       0x00       // Access disabled
-// #define SET_CCFG_CCFG_TAP_DAP_0_CPU_DAP_ENABLE          0xC5       // Access enabled if also enabled in FCFG
+// #define SET_CCFG_CCFG_TAP_DAP_0_CPU_DAP_ENABLE       0x00       // Access disabled
+#define SET_CCFG_CCFG_TAP_DAP_0_CPU_DAP_ENABLE          0xC5       // Access enabled if also enabled in FCFG
 #endif

 #ifndef SET_CCFG_CCFG_TAP_DAP_0_PRCM_TAP_ENABLE
@@ -238,8 +238,8 @@
 #endif

 #ifndef SET_CCFG_CCFG_TAP_DAP_0_TEST_TAP_ENABLE
-#define SET_CCFG_CCFG_TAP_DAP_0_TEST_TAP_ENABLE      0x00       // Access disabled
-// #define SET_CCFG_CCFG_TAP_DAP_0_TEST_TAP_ENABLE         0xC5       // Access enabled if also enabled in FCFG
+// #define SET_CCFG_CCFG_TAP_DAP_0_TEST_TAP_ENABLE      0x00       // Access disabled
+#define SET_CCFG_CCFG_TAP_DAP_0_TEST_TAP_ENABLE         0xC5       // Access enabled if also enabled in FCFG
 #endif

 #ifndef SET_CCFG_CCFG_TAP_DAP_1_PBIST2_TAP_ENABLE
```
After that register modification, we can flash the chip w/o having needed to do mass erase. 

So looks like there are some routine lockup on 4x4 chip that makes the serial output doesn't produce anything. I'm setting up a debugger...

henkyprayoga, 2017-02-20:
Found the problem. The code lock up during LF oscillators activation.

excerp cpu/cc26xx-cc13xx/dev/oscillators.c, func: oscillators_select_lf_xosc:

```c

  /* Switch LF clock source to the LF XOSC if required */
  if(ti_lib_osc_clock_source_get(OSC_SRC_CLK_LF) != OSC_XOSC_LF) {
    ti_lib_osc_clock_source_set(OSC_SRC_CLK_LF, OSC_XOSC_LF);

    /* Wait for LF clock source to become XOSC_LF */
    while(ti_lib_osc_clock_source_get(OSC_SRC_CLK_LF) != OSC_XOSC_LF); // IT KEEP LOOPING HERE

```
Seems like this procedure failed on 4x4...

g-oikonomou, 2017-02-20:
OK I think I may have an idea on this one.

Can you figure out which function call this is and where it is getting called?

henkyprayoga, 2017-02-20:
The call stack is:
1. contiki-main.c, func: int main(), oscillators_select_lf_xosc();
2. cpu/cc26xx-cc13xx/dev/oscillators.c, func: oscillators_select_lf_xosc();

Attached our schematic on the LF oscillators side, maybe it can help. We use TXCO.

[image]

henkyprayoga, 2017-02-20:
Seems like it's TI related issue. I also post a topic on TI support forum

g-oikonomou, 2017-02-20:
I know the LF RC OSC is buggy and must not be used, so we just select it when going to full shutdown (this is what I had in mind earlier). I'm not sure about the LF XOSC, not real experience with that but I'll ask someone I know to have a look at your schematic.

henkyprayoga, 2017-02-20:
I will also try to debug the hardware, check the 1.8v LDO, TXCO pulse, etc. It will need some time since it's very small. Its only 15x15mm in size.

[image]

henkyprayoga, 2017-02-20:
Owh, another thing. Since we are going to use 433MHz, we are using revision B (chip version 2.1). It may have different behaviour on the LF Osc side...

g-oikonomou, 2017-02-20:
I think different versions of CCxxware handle different PG versions differently if you know what I mean. I'd say make sure you are using very latest CCxxware. I bet TI may also be able to help you (in a sense of whether specific CCxxware versions support your chip)

henkyprayoga, 2017-02-20:
I can make the board running well by commenting the XOSC activation on contiki-main.c

```c
/*---------------------------------------------------------------------------*/
/**
 * \brief Main function for CC26xx-based platforms
 *
 * The same main() is used for all supported boards
 */
int
main(void)
{
  /* Enable flash cache and prefetch. */
 ti_lib_vims_mode_set(VIMS_BASE, VIMS_MODE_ENABLED);
 ti_lib_vims_configure(VIMS_BASE, true, true);

 ti_lib_int_master_disable();

  /* Set the LF XOSC as the LF system clock source */
 //..oscillators_select_lf_xosc();

  lpm_init();

  board_init();

  gpio_interrupt_init();
```

So the problem is on XOSC. I'm replacing CC13xxware using the latest version, still no joy. So possibly it hardware issue. So in general, remapping the pin on board.h and disabling LF XOSC could make contiki work on 4x4.

g-oikonomou, 2017-02-20:
Check the cc26xx errata, the lf rc osc should not be used. Not sure if the same applies to recent 13xx chip versions. I'm curious now

henkyprayoga, 2017-02-20:
After testing the board, I can assure that the LDO is outputing 1.789V and the TXCO outputing 38kHz on my scope. So looks like its CC1310 issue.

henkyprayoga, 2017-02-21:
Got explanation from TI's employee:

> A external 32 kHz clock should be connected to a DIO, not Q1.
>
> In the CCFG.c file you should set:
> #define SET_CCFG_MODE_CONF_SCLK_LF_OPTION 0x1 // External LF clock
> #define SET_CCFG_EXT_LF_CLK_DIO 0x01 // DIO number if using external LF clock
> 
> This is not well documented, I'm going to add a ticket to change the clock section of the TRM. 
> 
> The hardware is designed to detect if a crystal connected starts to swing but doesn't have circuitry to > detect if a TCXO has correct swing.

So my schematic is wrong, I should configure them as external clock, not as crystal replacement. 

Thank you for your great help so far.

closed 2017-02-21 by henkyprayoga

g-oikonomou, 2017-02-21:
No worries, please post if you manage to make things work

henkyprayoga, 2017-02-21:
At this moment I'm sticking with internal LF RC OSC. It seems to work pretty well on Rev. 2.1. I might encounter MAC issue later, since probably RC OSC is not accurate enough. I'm arranging 2nd sampling production with revised TCXO subsystem. Also, I have testing external LF clock functionality using external DS3231 chip and it works pretty well. So I guess the problem is really in my hardware design.

marciolm, 2017-12-28:
I had the same problem with the modules E70-915MS14 (cc1310 4x4), with no internal LF xtal. Commenting the line //..oscillators_select_lf_xosc(); on contiki-main.c the module started to work, but after a while, it stopped working until a reboot.The final solution was soldering a 32 kHz xtal on pins Q1 and Q2 of the module and two 15 pF capacitors between these pins and ground. That line in contiki-main was uncommented. 
Thanks for this thread, saved me a lot of time.
Márcio