contiki-os/contiki #2119: sensortag broker ip setting

JungSuWhan, 2017-03-03:
I use the cc26xx-web-demo source file on the ti cc2650 sensortag.
I have a mosquitto mqtt broker server. In the process of compiling the source
Changed broker ip, but it does not change normally.
Below the source code is the following comment:
"* If the node has a broker IP setting saved on flash, this value here will
  * Get ignored "

Changing the mqtt broker ip on a 6LBR web page takes too long.
How do I delete the brokerIP from Flash and compile it into a new IP?

tonnenpinguin, 2017-03-03:
Use the flash programmer 2 tool and perform a "forced mass erase" (you'll find it under the wrench symbol)

g-oikonomou, 2017-03-03:
> Use the flash programmer 2 tool and perform a "forced mass erase" (you'll find it under the wrench symbol)

This advice is inaccurate. The configuration is stored in external flash and this procedure will not erase external flash.

You can erase config from flash through the CoAP interface or the HTTP interface. You can also press the sensortag button (cannot remember which) for longer than 5 secs and this will also erase flash.

JungSuWhan, 2017-03-06:
Thank you for your help.
This problem has been solved by letting ext_flash_erase () function be executed when sensor tag button is pressed.

closed 2017-03-07 by g-oikonomou