contiki-os/contiki #2121: Executing msp430x instruction but mcu is not a msp430x

Shridharfly, 2017-03-05:
I am using sky motes in cooja. The simulation for the same code worked fine for first few times but now suddenly the following Contiki error:
`
 
    [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) - Executing MSP430X instruction but MCU is not a MSP430X  [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) - 
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) - Stack Trace: number of calls: 3 PC: $02076
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   process_thread_ctimer_process (local in ctimer.c) called from PC: $0674e (elapsed: 23678)
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   call_process (local in process.c) called from PC: $068f2 (elapsed: 23713)
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   process_run (memset.c) called from PC: $04228 (elapsed: 24975)
`
Java Stack Trace:
`

       [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) - org.contikios.cooja.ContikiError
      [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   at org.contikios.cooja.mspmote.MspMote.execute(MspMote.java:341)
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   at org.contikios.cooja.mspmote.MspMote.execute(MspMote.java:298)
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   at org.contikios.cooja.motes.AbstractWakeupMote$1.execute(AbstractWakeupMote.java:47)
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   at org.contikios.cooja.Simulation.run(Simulation.java:280)
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   at java.lang.Thread.run(Thread.java:745)
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) - Caused by: se.sics.mspsim.core.EmulationException: Executing MSP430X instruction but MCU is not a MSP430X
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   at se.sics.mspsim.core.MSP430Core.emulateOP(MSP430Core.java:1079)
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   at se.sics.mspsim.core.MSP430.stepMicros(MSP430.java:253)
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   at org.contikios.cooja.mspmote.MspMote.execute(MspMote.java:337)
     [java]  INFO [AWT-EventQueue-0] (MessageList.java:261) -   ... 4 more
`
I am working with latest version of contiki and gcc. I have tried restarting cooja after ant clean. Any idea on how do I solve the problem?

tlos, 2017-03-05:
Hello,

I used to get such errors when there was an error with my arrays. Maybe if
you check whether data is too long to fit in your array's limits or not,
you can solve this issue.

&ahmet.

Shridharfly, 2017-03-05:
I am using `rpl-udp` and have set `MAX_PAYLOAD_LEN` to 19. So data is definitely not too long to fit. And the simulation worked correctly for the same code earlier;  I have done not updates or upgrades after my code started working properly but suddenly starting throwing the above error.

Shridharfly, 2017-03-05:
Oh wait, you just solved it! THE DATA IS TOO SMALL .. 
(For future users) This issue occurs if the data is too long to fit in array or too small. 
@tlos Thanks a lot!

closed 2017-03-05 by Shridharfly

reubenct, 2019-01-29:
Hi, I too am facing the same error when I was testing my custom MAC protocol. I tried implementing ALOHA.
I observed that the number of system calls in the error changes when I vary `MAX_PAYLOAD_LEN`, but error is still persistent.
Please help me if you have any suggestions.